realsenseai/realsense-ros #917: Both D435 and T265 not working on Raspberry Pi 4

edgarcamilocamacho, 2019-09-07:
Platform: Raspberry Pi 4
OS: Raspbian Buster
Kernel: 4.19

Consider the following launchfiles:

``` xml
<launch>
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="cam_D"/>
    <arg name="serial_no" value="819112072640"/>
  </include>
</launch>
```

``` xml
<launch>
  <remap from="/cam_T/odom/sample" to="/odom" /> 
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="cam_T"/>
    <arg name="serial_no" value="925122110167"/>
  </include>
</launch>
```

If I launch them independently, the work fine, but when launch one of them when the other is running, I get the following error messages (in this case, I launched D435 first, and then T265):

```
21:20:14.241 [4815] [E] Device-2950: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
21:20:14.241 [4815] [E] Device-2950: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
.....
[ INFO] [1567818532.492638596]: Running with LibRealSense v2.28.0
 06/09 20:08:52,960 ERROR [2801652752] (types.h:304) Could not open device. Last Error: Device or resource busy
terminate called after throwing an instance of 'rs2::backend_error'
  what():  Could not open device. Last Error: Device or resource busy
[cam_T/realsense2_camera_manager-1] process has died [pid 13405, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager /cam_T/odom/sample:=/odom __name:=realsense2_camera_manager __log:=/home/pi/.ros/log/9fe20a46-d107-11e9-a3b8-3413e840e5cf/cam_T-realsense2_camera_manager-1.log].
log file: /home/pi/.ros/log/9fe20a46-d107-11e9-a3b8-3413e840e5cf/cam_T-realsense2_camera_manager-1*.log
```

If I launch them in the same launchfile, like this:
``` xml
<launch>

  <remap from="/cam_T/odom/sample" to="/odom" /> 
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="cam_T"/>
    <arg name="serial_no" value="925122110167"/>
  </include>

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="cam_D"/>
    <arg name="serial_no" value="819112072640"/>
  </include>

</launch>
```

Or this:
``` xml
<launch>
  <remap from="/cam_T/odom/sample" to="/odom" /> 
<include file="$(find realsense2_camera)/launch/rs_multiple_devices.launch">
    <arg name="serial_no_camera1" value="925122110167"/>
    <arg name="serial_no_camera2" value="819112072640"/>
  </include>

</launch>
```

I get the same errors.

MartyG-RealSense, 2019-09-07:
Apparently, if D435 is launched first then it can cause problems for T265.

https://github.com/IntelRealSense/realsense-ros/issues/774#issuecomment-493725159

A few comments down from that one, a community member suggested code for a fix to delay D435's launch in ROS.

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Thanks MartyG-RealSense for referencing the other thread.

@edgarcamilocamacho - Were you able to get past this with the suggestions from the comments provided in previous issue?

edgarcamilocamacho, 2019-10-30:
Effectively it works in that way, I am having other problems but they are not related with this. Thank you very much