realsenseai/realsense-ros #918: occupancy-mapping branch: no data in the /occupancy topic

whoobee, 2019-09-08:
Hello,

I am trying to run the occupancy live demo from the occupancy-mapping branch.
I am running the ROS nodes on a Jetson Nano, where I have a T265 and an D435 cameras connected.
1) I compiled and installed LRS 2.24 and recompiled the kernel to support the video output of the D435.
In realsense-vewer I can see that the cameras are running fine (even though some frames are dropped now and then).

2) I cloned the realsense-ros/occupancy-mapping repo and managed to run catkin_make without any issues.

3) I added the Serial-Numbers of the cameras in the `cameras.launch` and configured the nodlets to start `standalone` in order to avoid the issue with the T264-D435 startup.

4) I started the occupancy_live_rviz.launch launch file:
`roslaunch occupancy occupancy_live_rviz.launch`
and I can see the camera feeds and the point-cloud, but the map is not generated.

In RVIZ I can see the warning: `No map received`
[image]

When I run `rostopic echo /occupancy` there is no data echoed by the `/occupancy` topic

Am I missing something here?

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

Hi @whoobee

Apologies for no response for some time.  The occupancy-mapping demo was just that a demo of how to utilize the T265 tracking solution and a depth camera like D435 to show what can be done using both depth and tracking pose data.  There hasn't been any maintaining or updating of this example over time.

If anything else is needed, let us know or we can close out this thread.

closed 2020-02-21 by whoobee

nadaaqeel, 2022-07-06:
@whoobee Hello, Were you able to figure it out?

whoobee, 2022-07-07:
Hello @, sadly no. The occupancy branch is verry old and not maintaned anymore. It is also incompattible with the newer versions of Realsense library. So I gave up on this idea 😀