realsenseai/realsense-ros #920: occupancy-mapping

jungladicitta, 2019-09-10:
what does those parameters means?
```
  <param name="HeightOfInterestMin" value="-0.5"/>
  <param name="HeightOfInterestMax" value="0.5"/>
  <param name="DepthOfInterestMin" value="0.2"/>
  <param name="DepthOfInterestMax" value="2.0"/>
```
in occupancy.launch file?

jungladicitta, 2019-09-16:
anyone? @schmidtp1?

schmidtp1, 2019-09-16:
@jungladicitta, hi, this is the space above and below the camera that is considered when determining if the space is occupied. This depends on where the camera is mounted on your robot and the height of the robot, for example min. is the height above ground (including some margin) and max. is the height of the robot minus height of the camera (plus some margin).
Depth of interested is used to clip the depth. There is a minimum distance in which the camera returns valid values ("minZ", here set to 20cm). As with any stereo system there depth error grows with the distance (quadratically) and the depth can be limited to bound the error (here to 2m).

jungladicitta, 2019-09-19:
@schmidtp1 that is very interesting and good to know!
thanks for reply

jungladicitta, 2019-09-19:
@schmidtp1 why is there "-0.5" in HeightOfInterestMin? does it mean 50cm below the ground?
Is this because we need margin due to d435 error in 2 meters?

jungladicitta, 2019-09-19:
@schmidtp1 so, "-0.5" in HeightOfInterestMin means that our camera is located 50cm above the ground, right?

schmidtp1, 2019-09-19:
@jungladicitta, correct!

jungladicitta, 2019-09-20:
@schmidtp1 do I need something to take into account if locate camera angled to the floor, like this
[image]

jungladicitta, 2019-09-20:
@schmidtp1 dont you think its vice versa? i tried as you said, and occupancyGrid become worse, there is no even 1 meter tables

schmidtp1, 2019-09-20:
@jungladicitta, as far as I remember, the height refers to the to the T265 reference frame.
The point cloud is transformed from the depth camera to the T265 frame using the extrinsics here: https://github.com/IntelRealSense/realsense-ros/blob/occupancy-mapping/realsense2_camera/urdf/mount_t265_d435.urdf.xacro
Then it is transformed into a world reference system using the pose estimates from T265. This reference system is initialized at the position of the T265 at startup. While the frame is gravity aligned, there is no reference for the height of the camera, i.e. it is set to 0. Everything should be expressed with respect to the (initial) position/height of the T265 thereby. Please refer to the T265 data sheet for details about the coordinate frames and a more thorough description.

schmidtp1, 2019-09-20:
> @schmidtp1 do I need something to take into account if i locate camera with inclination to the floor?
> like this
> [image]

Where is your T265 mounted?

jungladicitta, 2019-09-23:
@schmidtp1 I haven't changed that numbers. Here is part of my xacro file:
```
  <!-- REALSENSE D435 -->
  <joint name="cameras_joint" type="fixed">
    <parent link="base_footprint"/>
    <child link="t265_pose_frame"/>
    <origin xyz="0.0 0.0 1.2985" rpy="0 0 0"/>
  </joint>

  <link name="t265_pose_frame"/>
  <link name="d400_link"/>

  <joint name="t265_to_d400" type="fixed">
    <parent link="t265_pose_frame"/>
    <child link="d400_link"/>
    <origin xyz="0.009 0.021 0.027" rpy="0.000 -0.018 0.005"/>
  </joint>
```

I am also using static_transform publishers in launch file. Here is my full launch file:
```
<launch>

  <!-- CAMERA -->
  <include file="$(find turtlebot3_occupancy_mapping)/launch/robot_and_cameras.launch"/>

  <!-- occupancy-mapping -->
  <include file="$(find turtlebot3_occupancy_mapping)/launch/occupancy.launch">
  </include>

  <node pkg="turtlebot3_occupancy_mapping" type="map_frame_publisher.py" name="map_frame_publisher" output="screen">
  </node>

  <node name="static_tf0" pkg="tf2_ros" type="static_transform_publisher" args="0 0 1.2985 0 0 0 /map /t265_odom_frame"/>

  <node name="static_tf1" pkg="tf2_ros" type="static_transform_publisher" args="0 0 0 0 0 0 /map /odom"/>

</launch>
```
where map_frame_publisher.py is just publishing occupancy topic to map frame:
```
#!/usr/bin/env python  
import rospy

# Because of transformations
import tf_conversions

import tf2_ros
import tf2_msgs.msg
import geometry_msgs.msg
import sensor_msgs.msg
import nav_msgs.msg
from std_msgs.msg import String

def handle_map_frame(data):
    pub = rospy.Publisher("map", nav_msgs.msg.OccupancyGrid, queue_size=10)
    rate = rospy.Rate(10) # 10hz

    data.header.frame_id = "map"

    pub.publish(data)
    rate.sleep()

if __name__ == '__main__':
    rospy.init_node('map_frame_broadcaster')
    rospy.Subscriber("/occupancy", nav_msgs.msg.OccupancyGrid, handle_map_frame)
    ##rospy.Subscriber('/sensor_us', sensor_msgs.msg.Range, handle_camera_pose)
    rospy.spin()
```
Am i doing something wrong?

jungladicitta, 2019-09-24:
> > @schmidtp1 do I need something to take into account if i locate camera with inclination to the floor?
> > like this
> > [image]
> 
> Where is your T265 mounted? You need to take it into account in the extrinsics in the link above.

T265 was mounted on the mast, 1.29 meters above the ground, oriented towards to ground.

But now I decided to move T265 at mid-height, 0.92 meters from the ground, oriented straight-ahead

jungladicitta, 2019-09-24:
@schmidtp1 
these numbers is only for d435 position relatively to t265, so there is no need to change them if I move cameras on the robot, right?
`<origin xyz="0.009 0.021 0.027" rpy="0.000 -0.018 0.005"/>`

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@jungladicitta is there anything further needed on this thread?

closed 2020-01-03 by RealSenseCustomerSupport