realsenseai/realsense-ros #922: Cannot lauch d435 camera node

catmasteryip, 2019-09-11:
I am using Ubuntu 18.04.3 LTS
kernel version: 5.0.0-27-generic
ROS version: melodic
Intel realsense-ros SDK version: =>2.0
When I entered `roslaunch realsense2_camera rs_camera.launch` in bash after installation of Intel realsense-ros  SDK, this happened:
[image]
`11/09 12:05:41,072 WARNING [140710628022016] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.`

catmasteryip, 2019-09-11:
D435i firmware version: 05.11.11.100

catmasteryip, 2019-09-23:
Thanks for the reply. I have reinstalled ROS and librealsense ROS wrapper but the problem persists.

catmasteryip, 2019-09-23:
I am currently using ROS melodic on Ubuntu 18 but not the ROS kinetic on Ubuntu 16 as suggested, would that be a problem?

catmasteryip, 2019-09-23:
[https://github.com/IntelRealSense/realsense-ros/issues/492#issuecomment-533528194](url)
It seems to me that the camera node can publish but the roslaunch command says otherwise.

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

@catmasteryip Which librealsense and ROS wrapper version did you use when get the issue? Did you get chance to try the latest version? We tried Ubuntu18.04+melodic here but didn't get your issue.

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@catmasteryip Sorry that we didn't get any response from you for weeks, will close the issue at this point. Please feel free to create another new one if you still have questions or issues. Thanks!

closed 2019-12-13 by RealSenseCustomerSupport