realsenseai/realsense-ros #926: Settings and filters for d435 as input for SLAM

Joar93, 2019-09-17:
Hello!

i'm currently using the t265 and d435 along with lidar sensors to run 3D SLAM with cartographer.  I have played around with several different post processing filters and control settings and my question is does anyone have any good examples of settings to use for a SLAM application? The SLAM is 3D however the application is driving on the floor at all times. I want the pointcloud2 from the d435 to have a high accuracy rather then high density. I know this question is very general buy any help is appriciated!

Warm regards
Joar

MartyG-RealSense, 2019-09-18:
The 400 Series cameras have 'Visual Preset' configurations that balance the camera settings towards certain focuses such as high accuracy or high density.  You can also create your own custom preset file by configuring settings in the RealSense Viewer software and exporting them as a 'json' preset file.  Have you tried the High Accuracy preset please?

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets

In regard to laser power, the camera has temperature protections.  Once temperature reaches an excess level for a certain period of time, laser power is halved.  If the high temperature continues for a further period of time after that then the camera shuts off.

Advice about laser lifetime is provided in the link below:

https://github.com/IntelRealSense/librealsense/issues/3452#issuecomment-471492320

If you anticipate high sustained temperatures, the new FRAMOS D445e can operate at 55 degrees instead of the Intel D435's 35 degree recommended limit.

https://github.com/IntelRealSense/librealsense/issues/4769

Joar93, 2019-09-18:
Hello @MartyG-RealSense !

First of wow! Amazing and quick feedback, thank you!
Yes! After a couple of days of testing i found out that the High accuracy preset works best for our SLAM solution! (Cartographer 3D with depth camera as input) It does not create very detailed maps but it tracks the robots location pritty well). I also used the filters settings that was used in the high accuracy preset.

Thank you for the information regarding the laser! That kind of info was exactly what i needed! A hard question but in your experience is the d435 made for operating around 12 hours a day every day or is that workload to excessive?

Warm regards!
Joar

MartyG-RealSense, 2019-09-18:
You are very welcome.  :)

The ability for the 400 Series cameras to run for long periods is addressed in a comment above the one about laser lifespan.

https://github.com/IntelRealSense/librealsense/issues/3452#issuecomment-471455156

Joar93, 2019-09-18:
Thanks for the quick feedback!
I will take a look at the MIT project, looks very interesting!

Warm regards
Joar

closed 2019-09-18 by Joar93

504115681, 2021-09-25:
> Im currently running at the recommended resolution of 800*480 with a decimation, spatial and temporal filter!

Hello, how do you set this up?

MartyG-RealSense, 2021-09-25:
Hi @504115681   The links below have examples of code in C++ and Python programming languages for setting up these post-processing filters.

**C++**
https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing
https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp

**Python**
https://github.com/IntelRealSense/librealsense/issues/1672#issuecomment-387438447

You can also apply post-processing filters in the **RealSense Viewer** tool by expanding open the **Stereo Module** options in the Viewer's side panel and going to the **Post-Processing** options sub-category.