realsenseai/realsense-ros #928: How to set the origin of the coordinate system when modeling with two sensors？

TomLuosy, 2019-09-17:
I used two d435 sensors for rotational modeling. The center axis of rotation is in the middle of the two sensors, but my modeling effect has been poor, no matter how I adjust the translation matrix T and the rotation matrix R in the camera matrix. Later, I found in the official document that the origin of the reference coordinate system is set to the left sensor. How do I set this point between the two sensors to improve the rotation modeling effect?
[image]

doronhi, 2019-09-25:
Depth is calculated with respect to the left IR sensor. Therefor the reference coordinate system is set to the left sensor.
You can add another TF describing the transformation from camera_link (left IR sensor) to the mounting point or to the center of the camera, as suits you best. You may find the schematics in page 32 [here](https://www.mouser.com/pdfdocs/Intel_D400_Series_Datasheet.pdf) useful.

RealSenseCustomerSupport, 2019-11-08:
----------------------------------------------

Hi @TomLuosy,

Any update? I believe the info from @doronhi can resolve your issue, right?
You can also refer to the latest datasheet rev007.
https://www.intelrealsense.com/wp-content/uploads/2019/10/Intel-RealSense-D400-Series-Datasheet-Oct-2019.pdf?_ga=2.113960391.1765568705.1573001324-1515688634.1554101320

closed 2019-12-06 by RealSenseCustomerSupport