realsenseai/realsense-ros #929: Support D435i non rectified raw infrared image at 1280x800

yliupub, 2019-09-18:
Hi, 

Just want to check if it is possible to support the none rectified infrared image output at 1280x800 @ 25fps. Check with the current code, it is always the rectified image in case of infrared output ros topic.  

Thanks,

Yang

doronhi, 2019-09-25:
Depth is available only with the rectified infrared images. The whole node is built around the depth image so your request is not on the table right now.
Would you like to elaborate what is your need?

RealSenseCustomerSupport, 2019-11-08:
----------------------------------------------

@yliupub Any update for doronhi's question? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@yliupub Sorry that we didn't get update from you for weeks. Will close the ticket at this point. Please create another one if you still have questions/issues. Thanks!

closed 2019-12-02 by RealSenseCustomerSupport