realsenseai/realsense-ros #931: D435&T265 serial needed?

gbr1, 2019-09-20:
Hi,
just working on T265 and D435 but strange errors appeared, such as D435 considered a T265.
Is it needed serial number of each device as shown here [rs_d400_and_t265.launch](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_d400_and_t265.launch)?

Using **realsense-viewer**, it can automatically detect which one is a D435 or a T265.
Is it impossible in realsense ros wrapper?

Thank you in advance!

MartyG-RealSense, 2019-09-21:
If you are using a D435 and T265 together, you should make sure that the D435 is not activated first, otherwise it can cause problems.  The link below has a script for delaying the D435's launch in ROS.

https://github.com/IntelRealSense/realsense-ros/issues/774#issuecomment-493933596

gbr1, 2019-09-23:
Hi @MartyG-RealSense !

Thank you for your suggestion.

I edit using timed_roslaunch:
[https://github.com/gbr1/erwhi-hedgehog-ros/blob/kinetic-devel-t265-support/erwhi_bringup/launch/include/sensors.launch#L27](https://github.com/gbr1/erwhi-hedgehog-ros/blob/kinetic-devel-t265-support/erwhi_bringup/launch/include/sensors.launch#L27)
but still doesn't work :(
D435 is launched as T265 node.
I would like to not add serial number in nodes to simplify user experience.

Thank you in advance.

MartyG-RealSense, 2019-09-23:
Another user got around the problem of needing a serial number by instead programming query_devices to look for the first device with the name 'T265' in it.

https://github.com/IntelRealSense/librealsense/issues/3434#issuecomment-476290706

doronhi, 2019-09-25:
Hi @gbr1, 
I understand the annoyance in setting the serial numbers but I'm not sure I understand your proposal.
In "rs_d400_and_t265.launch" you set two nodes with parameters in each node like fps etc.
Two node start running. Do you mean that a node should know if it runs T265 or D435?
Do you mean that there should be a parameter like device_type? And serial number will only take affect if more than one device of a given type is found?

P.S.
I think timed_roslaunch is not needed and there is no longer an issue with which node starts first.

gbr1, 2019-09-25:
Hi @doronhi!

Yes, the problem is to check on each robot the serial number.

If I don't put it in arguments when I run T265 node, D435 starts instead.
If I detach D435 usb, T265 works correctly.

If I use realsense-viewer, everything is ok.  Devices are correctly detected.
I'm using the release 2.28.0 and latest commit of realsense ros.

Thank you in advance.

doronhi, 2019-09-25:
I understand the need. I think it will help many other users. It was not on my desk so it might take me a while.

gbr1, 2019-09-25:
@doronhi just tested that only T265 serial is needed:
[https://github.com/gbr1/erwhi-hedgehog-ros/blob/kinetic-devel-t265-support/erwhi_bringup/launch/include/sensors.launch#L28](https://github.com/gbr1/erwhi-hedgehog-ros/blob/kinetic-devel-t265-support/erwhi_bringup/launch/include/sensors.launch#L28)

So I will put only it for the moment!

BriceRenaudeau, 2019-10-08:
Personally, to use both T265 and D435 together I exported the serial as environment variables :
```
export T265_SERIAL=$(rs-enumerate-devices | grep -A1 'T265' | grep -v "T265" | grep 'Serial Number                 :' | cut -d ':' -f 2 | cut -d $'\t' -f 2 | cut -d $'\n' -f 1)
export D435_SERIAL=$(rs-enumerate-devices | grep -A1 '435' | grep -v "435" | grep 'Serial Number                 :' | cut -d ':' -f 2 | cut -d $'\t' -f 2 | cut -d $'\n' -f 1)
```
It works fine, you just need to open a new terminal when changing devices.

BriceRenaudeau, 2019-10-08:
In  the rs-enumerate-devices, the D435 is always on top(first to be detected?), I think that's why you only need the serial for the t265

doronhi, 2019-10-24:
Option added.

MartyG-RealSense, 2019-10-24:
@doronhi   Thanks so much for adding that - I will direct people to the new information.

doronhi, 2019-10-24:
> @doronhi Thanks so much for adding that - I will direct people to the new information.

Excellent! Thanks a lot.

RealSenseCustomerSupport, 2019-11-14:
----------------------------------------------

I see that some PRs went in to take care of this issue.  I want to make sure there isn't anything further needed here @gbr1.  If everything is ok then we can close this out.

closed 2019-12-05 by RealSenseCustomerSupport