realsenseai/realsense-ros #932: UVC metadata payloads not available; Unable to

catmasteryip, 2019-09-23:
When I ran `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`, rviz cannot render the video feed but camera node is publishing.
UVC metadata payloads issue is still unresolved. Uncertain if this issue relates to failure in rendering RGBD point cloud with rviz.
[image]
[image]

doronhi, 2019-09-25:
rviz states that "map" frame does not exist. Try changing the `fixed frame` option in the rviz panel from "map" to "camera_link".

doronhi, 2019-09-25:
regarding the "Frame metadata isn't available" message - what kernel version do you have?

FrederikHaa, 2019-10-02:
I get the same warning on kernel version 4.15, 4.18 and 5.0

doronhi, 2019-10-06:
> I get the same warning on kernel version 4.15, 4.18 and 5.0

The message about the metadata being unavailable is usually related to installation issues of librealsense. Did you install librealsense from code or with a debian package? Did you follow the installation procedure accordingly? (specifically, you need dkms package for installation from debian, see [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)  or a patch for kernel script if installed from code, see [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) )? Be sure not to mix these 2 procedures.

FrederikHaa, 2019-10-09:
That might be it.

I've installed librealsense with the debian packages, but when I updated to linux 5.0 it did not work anymore, so I tried to patch the kernel manually. After an update of librealsense the ROS package worked again, but i got the UVC metadata errors.

How do i uninstall the manual patches?

Grandzxw, 2019-11-01:
> 可能就是这样。
> 
> 我已经用debian软件包安装了librealsense，但是当我更新到linux 5.0时，它不再起作用了，所以我尝试手动修补内核。在更新librealsense之后，ROS软件包再次起作用，但是我收到了UVC元数据错误。
> 
> 如何卸载手动补丁？

have you solved it?

Grandzxw, 2019-11-08:
> The message about the metadata being unavailable is usually related to installation issues of librealsense. Did you install librealsense from code or with a debian package? Did you follow the [...]

When I 

> The message about the metadata being unavailable is usually related to installation issues of librealsense. Did you install librealsense from code or with a debian package? Did you follow the [...]

 **when I run roslaunch realsense2_camera rs_camera.launch, it happens:**

[ INFO] [1573206763.994774042]: Initializing nodelet with 8 worker threads.
[ INFO] [1573206764.047444526]: RealSense ROS v2.2.9
[ INFO] [1573206764.047473041]: Running with LibRealSense v2.29.0
[ INFO] [1573206764.593602846]:
[ INFO] [1573206764.625195138]: Device with serial number 909512072035 was found.

[ INFO] [1573206764.625311882]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video1 was found.
[ INFO] [1573206764.625400208]: Device with name Intel RealSense D435I was found.
[ INFO] [1573206764.631214316]: Device with port number 2-3 was found.
[ INFO] [1573206764.643706928]: getParameters...
[ INFO] [1573206764.710479104]: setupDevice...
[ INFO] [1573206764.710499423]: JSON file is not provided
[ INFO] [1573206764.710527300]: ROS Node Namespace: camera
[ INFO] [1573206764.710552986]: Device Name: Intel RealSense D435I
[ INFO] [1573206764.710586034]: Device Serial No: 909512072035
[ INFO] [1573206764.710626285]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video1
[ INFO] [1573206764.710644769]: Device FW version: 05.11.15.00
[ INFO] [1573206764.710662772]: Device Product ID: 0x0B3A
[ INFO] [1573206764.710677258]: Enable PointCloud: Off
[ INFO] [1573206764.710692892]: Align Depth: Off
[ INFO] [1573206764.710707502]: Sync Mode: Off
[ INFO] [1573206764.710926370]: Device Sensors:
[ INFO] [1573206764.711103725]: Stereo Module was found.
[ INFO] [1573206764.711156824]: RGB Camera was found.
[ INFO] [1573206764.711189067]: Motion Module was found.
[ INFO] [1573206764.711257667]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1573206764.711291139]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1573206764.711300801]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1573206764.711318994]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1573206764.711455362]: num_filters: 0
[ INFO] [1573206764.711467706]: Setting Dynamic reconfig parameters.
[ INFO] [1573206764.863620782]: Done Setting Dynamic reconfig parameters.
[ INFO] [1573206764.881246288]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1573206764.882123827]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1573206764.882906152]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
08/11 17:52:44,885 WARNING [140046886799104] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1573206764.902568590]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1573206764.903316010]: setupPublishers...
[ INFO] [1573206764.907290857]: Expected frequency for depth = 30.00000
[ INFO] [1573206764.934100284]: Expected frequency for infra1 = 30.00000
[ INFO] [1573206764.958084388]: Expected frequency for infra2 = 30.00000
[ INFO] [1573206764.981351051]: Expected frequency for color = 30.00000
[ INFO] [1573206765.003662835]: setupStreams...
[ INFO] [1573206765.004525689]: insert Depth to Stereo Module
[ INFO] [1573206765.004660836]: insert Color to RGB Camera
[ INFO] [1573206765.004696125]: insert Infrared to Stereo Module
[ INFO] [1573206765.004726470]: insert Infrared to Stereo Module
[ INFO] [1573206765.004746949]: insert Gyro to Motion Module
[ INFO] [1573206765.004768833]: insert Accel to Motion Module
[ INFO] [1573206765.137295673]: SELECTED BASE:Depth, 0
08/11 17:52:45,158 WARNING [140046551557888] (sensor.cpp:1152) HID timestamp not found, switching to Host timestamps.
08/11 17:52:45,191 WARNING [140046199944960] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ INFO] [1573206765.222268005]: RealSense Node Is Up!
08/11 17:52:45,257 WARNING [140046585128704] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.

RealSenseCustomerSupport, 2019-12-12:
----------------------------------------------

@Grandzxw Did you get any error when install debian package? UVC metadata error might be caused incomplete installation. Please double check. Thanks!

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@Grandzxw Did you get chance to check? Any update from your side? Looking forward to your update. Thanks!

BriceRenaudeau, 2020-01-13:
Hello, I hade the same problem :
WARNING [...] (ds5-timestamp.cpp:64) UVC metadata payloads not available
[ WARN] [/camera_right/realsense2_camera_manager]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)

[Ubuntu 18.04, kernel 5.0, melodic, D435, firmware 5.12, librealsense2-dkms up-to-date]

I hade no issues until today where I found that the TimeStamp in the depth images published by my camera where stuck in the past.
The stamps were oscillating in the past (not fixed).
Strangely when I added `<arg name="filters" default="pointcloud"/>` in the launch file, the stamps were good again.

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

@BriceRenaudeau Please refer to the comment below to download DKMS for kernel 5.0 to try again.

https://github.com/IntelRealSense/librealsense/issues/4586#issuecomment-583046210

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@BriceRenaudeau Did you get chance to try? Could you please update? Thanks!

BriceRenaudeau, 2020-02-14:
Hi, Thanks for your answer.
I was working around with "pointcloud" filter option. 
After the latest update, I can do it the normal way :+1: 

But Strangely, the DKMS for 5.0 kernel broke the PR I made to be able to connect to devices using the usb physical ports. (physical port increment every time I plug/unplug the D435)
Also, my launch for D435 doesn't work either.  Hence I'm still using the old one.

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@BriceRenaudeau Thanks for the quick update! Please feel free to create another new ticket to track the DKMS issue.

RealSenseCustomerSupport, 2020-02-27:
----------------------------------------------

@BriceRenaudeau As the issue resolved, will close this accordingly. Please create another ticket to track the DKMS issue. Thanks!

closed 2020-02-27 by RealSenseCustomerSupport