realsenseai/realsense-ros #933: How to improve localization in differential robot with D435 and T265 camera WITHOUT wheel odometry.

SSusantAchary, 2019-09-26:
I have a differential robot which has D435 and T265 realsense camera's mounted on it.
Issue:
On some occasions the robot seems to move inside the area(map is generated before navigation using gmapping, depthimage_to_laserscan packages) of obstacle region(static) during navigation. and i have no wheeler odometry(no feedback from wheels) data for T265 to optimize on localization.

How to improve localization without wheel odometry? 
![Uploading odom.png…]

The odometer block is missing in my case.

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

Hi @SSusantAchary

I'd like to suggest a few things if you haven't done so already.  Please upgrade to the latest librealsense package 2.31.0.  Have a look at the release notes, there has been some updates to the release with regards to localization and other fixes.  Environment also plays a factor in identifying features in a room for the SLAM algo that is run on the T265.  There is also some discussion on github regarding connecting both the D435 and T265 camera together:  https://github.com/IntelRealSense/librealsense/tree/master/examples/tracking-and-depth

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

Were you able to test out the latest releases?  If there is nothing more needed on this thread, we can close it.

closed 2020-03-17 by RealSenseCustomerSupport

steflayanto, 2020-06-22:
After watching the talk where the T265 was integrated with a D435 for occupancy mapping (at this [link](https://www.youtube.com/watch?v=62vm0_RZ1nU&feature=youtu.be&t=1036)), it appears that the D435 is **not** used to improve localization. It is used to produce an occupancy map but assumes accurate localization from the T265, and makes no improvement to pose estimation.

Going back to the original question, is it possible to improve **localization** specifically by integrating additional sensors such as LiDAR or depth cameras with the T265, similar to how wheeled odometry input is integrated as of now?

dschori, 2020-10-26:
I would also be interested in an answer to this question. Especially in combination with a LiDAR sensor.

doronhi, 2020-10-27:
As the T265's interface is not likely to change, I believe the best approach would be to use the T265's output, mainly the linear and radial velocities, and use some other SLAM node to integrate it with other sensors.

SSusantAchary, 2020-11-30:
Thank you Everyone, I approached the problem in a different way and solved it.

jameszw001, 2021-02-05:
> Thank you Everyone, I approached the problem in a different way and solved it.

If you don't mind, could you like to share what your approach is?