realsenseai/realsense-ros #934: Crash after running for several hours

VorpalBlade, 2019-09-26:
After running for a few hour the nodelet process crashes with an error about "corrupted size vs. prev_size". This has happened several times now.

The packages were installed using the repository of prebuilt packages for Ubuntu 16.04 and ROS Kinetic. (i.e. https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages and then installing ros-kinetic-librealsense2).

```
*** Error in `/opt/ros/kinetic/lib/nodelet/nodelet': corrupted size vs. prev_size: 0x00007f29103a3490 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f2954c8c7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x7e9dc)[0x7f2954c939dc]
/lib/x86_64-linux-gnu/libc.so.6(+0x80678)[0x7f2954c95678]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f2954c9953c]
/opt/ros/kinetic/lib/libroscpp.so(+0x9965a)[0x7f295609565a]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros10Connection14writeTransportEv+0x685)[0x7f2956097785]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros12TransportTCP12socketUpdateEi+0x18c)[0x7f295610f81c]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros7PollSet6updateEi+0x320)[0x7f295614c4f0]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros11PollManager10threadFuncEv+0x35)[0x7f29560d1ac5]
/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0(+0x115d5)[0x7f295419f5d5]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7f2953f786ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f2954d1c41d]
======= Memory map: ========
00400000-0040e000 r-xp 00000000 08:02 8128550                            /opt/ros/kinetic/lib/nodelet/nodelet
0060d000-0060e000 r--p 0000d000 08:02 8128550                            /opt/ros/kinetic/lib/nodelet/nodelet
0060e000-0060f000 rw-p 0000e000 08:02 8128550                            /opt/ros/kinetic/lib/nodelet/nodelet
0219f000-02694000 rw-p 00000000 00:00 0                                  [heap]
7f2900000000-7f2900021000 rw-p 00000000 00:00 0
7f2900021000-7f2904000000 ---p 00000000 00:00 0
7f2908000000-7f2908bf4000 rw-p 00000000 00:00 0
7f2908bf4000-7f290c000000 ---p 00000000 00:00 0
7f290c000000-7f290c021000 rw-p 00000000 00:00 0
7f290c021000-7f2910000000 ---p 00000000 00:00 0
7f2910000000-7f2910e5b000 rw-p 00000000 00:00 0
7f2910e5b000-7f2914000000 ---p 00000000 00:00 0
7f2914000000-7f2914aea000 rw-p 00000000 00:00 0
7f2914aea000-7f2918000000 ---p 00000000 00:00 0
7f2918000000-7f2918021000 rw-p 00000000 00:00 0
7f2918021000-7f291c000000 ---p 00000000 00:00 0
7f291c000000-7f291c031000 rw-p 00000000 00:00 0
7f291c031000-7f2920000000 ---p 00000000 00:00 0
7f29209ff000-7f292131d000 rw-p 00000000 00:00 0
7f29214e1000-7f2921579000 rw-p 00000000 00:00 0
7f292165b000-7f292250b000 rw-p 00000000 00:00 0
7f292250b000-7f292250c000 ---p 00000000 00:00 0
7f292250c000-7f2922d0c000 rwxp 00000000 00:00 0
7f2922d0c000-7f29232f2000 rw-p 00000000 00:00 0
7f29232f2000-7f29232f9000 r-xp 00000000 08:02 11019044                   /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7f29232f9000-7f29234f9000 ---p 00007000 08:02 11019044                   /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7f29234f9000-7f29234fa000 r--p 00007000 08:02 11019044                   /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7f29234fa000-7f29234fb000 rw-p 00008000 08:02 11019044                   /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7f29234fb000-7f2923514000 r-xp 00000000 08:02 11019305                   /usr/lib/x86_64-linux-gnu/libtheoradec.so.1.1.4
7f2923514000-7f2923713000 ---p 00019000 08:02 11019305                   /usr/lib/x86_64-linux-gnu/libtheoradec.so.1.1.4
7f2923713000-7f2923714000 r--p 00018000 08:02 11019305                   /usr/lib/x86_64-linux-gnu/libtheoradec.so.1.1.4
7f2923714000-7f2923715000 rw-p 00019000 08:02 11019305                   /usr/lib/x86_64-linux-gnu/libtheoradec.so.1.1.4
7f2923715000-7f2923753000 r-xp 00000000 08:02 11019307                   /usr/lib/x86_64-linux-gnu/libtheoraenc.so.1.1.2
7f2923753000-7f2923952000 ---p 0003e000 08:02 11019307                   /usr/lib/x86_64-linux-gnu/libtheoraenc.so.1.1.2
7f2923952000-7f2923953000 r--p 0003d000 08:02 11019307                   /usr/lib/x86_64-linux-gnu/libtheoraenc.so.1.1.2
7f2923953000-7f2923954000 rw-p 0003e000 08:02 11019307                   /usr/lib/x86_64-linux-gnu/libtheoraenc.so.1.1.2
7f2923954000-7f2923996000 r-xp 00000000 08:02 8131862                    /opt/ros/kinetic/lib/libtheora_image_transport.so
7f2923996000-7f2923b95000 ---p 00042000 08:02 8131862                    /opt/ros/kinetic/lib/libtheora_image_transport.so
7f2923b95000-7f2923b97000 r--p 00041000 08:02 8131862                    /opt/ros/kinetic/lib/libtheora_image_transport.so
7f2923b97000-7f2923b98000 rw-p 00043000 08:02 8131862                    /opt/ros/kinetic/lib/libtheora_image_transport.so
7f2923b98000-7f2923b9a000 rw-p 00000000 00:00 0
7f2923b9a000-7f2923bbb000 r-xp 00000000 08:02 8128646                    /opt/ros/kinetic/lib/libimage_transport_plugins.so
7f2923bbb000-7f2923dbb000 ---p 00021000 08:02 8128646                    /opt/ros/kinetic/lib/libimage_transport_plugins.so
7f2923dbb000-7f2923dbc000 r--p 00021000 08:02 8128646                    /opt/ros/kinetic/lib/libimage_transport_plugins.so
7f2923dbc000-7f2923dbd000 rw-p 00022000 08:02 8128646                    /opt/ros/kinetic/lib/libimage_transport_plugins.so
7f2923dbd000-7f2923dfc000 r-xp 00000000 08:02 8130142                    /opt/ros/kinetic/lib/libcompressed_image_transport.so
7f2923dfc000-7f2923ffc000 ---p 0003f000 08:02 8130142                    /opt/ros/kinetic/lib/libcompressed_image_transport.so
7f2923ffc000-7f2923ffe000 r--p 0003f000 08:02 8130142                    /opt/ros/kinetic/lib/libcompressed_image_transport.so
7f2923ffe000-7f2923fff000 rw-p 00041000 08:02 8130142                    /opt/ros/kinetic/lib/libcompressed_image_transport.so
7f2923fff000-7f2924000000 rw-p 00000000 00:00 0
[... 358 lines cut ...]
7fffce42f000-7fffce450000 rwxp 00000000 00:00 0                          [stack]
7fffce450000-7fffce451000 rw-p 00000000 00:00 0
7fffce541000-7fffce544000 r--p 00000000 00:00 0                          [vvar]
7fffce544000-7fffce546000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
[sensors/camera/realsense2_camera_manager-1] process has died [pid 12463, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/arvid/.ros/log/
4c6e600c-e044-11e9-bf7a-94c6911211bd/sensors-camera-realsense2_camera_manager-1.log].
log file: /home/arvid/.ros/log/4c6e600c-e044-11e9-bf7a-94c6911211bd/sensors-camera-realsense2_camera_manager-1*.log
[sensors/camera/realsense2_camera-2] process has finished cleanly
log file: /home/arvid/.ros/log/4c6e600c-e044-11e9-bf7a-94c6911211bd/sensors-camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
```

Of the mentioned log files, only the second exists, and it is very brief and with no useful information:
```
$ cat /home/arvid/.ros/log/4c6e600c-e044-11e9-bf7a-94c6911211bd/sensors-camera-realsense2_camera-2-stdout.log
[ INFO] [1569491963.810708023]: Loading nodelet /sensors/camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
[ INFO] [1569491963.861401992]: waitForService: Service [/sensors/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1569491963.997494320]: waitForService: Service [/sensors/camera/realsense2_camera_manager/load_nodelet] is now available.
[ INFO] [1569502968.395424261]: Bond broken, exiting
```

The only useful information that can be gleaned from this it that it was running for at least 3 hours before it happened.

VorpalBlade, 2019-09-26:
When I restarted the camera after this there were some unusual warning and error messages that doesn't show up on normal camera launch, this may be relevant information:
```
[ INFO] [1569505113.620459742]: SELECTED BASE:Depth, 0
[ INFO] [1569505113.723543321]: RealSense Node Is Up!
[ INFO] [1569505113.723812443]: Setting Dynamic reconfig parameters.
[ INFO] [1569505113.854171242]: Done Setting Dynamic reconfig parameters.
[ WARN] [1569505115.505813151]: No stream match for pointcloud chosen texture Process - Color
[ WARN] [1569505116.157496282]: Hardware Notification:Depth stream start failure,1.56951e+12,Error,Hardware Error
[ERROR] [1569505116.157600642]: Hardware Reset is needed. use option: initial_reset:=true
 26/09 15:38:37,265 ERROR [140622228862720] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 26/09 15:38:37,270 ERROR [140622228862720] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 26/09 15:38:37,304 ERROR [140622254040832] (types.h:304) xioctl(VIDIOC_DQBUF) failed for fd: 37 Last Error: No such device
 26/09 15:38:37,304 ERROR [140622126360320] (types.h:304) xioctl(VIDIOC_DQBUF) failed for fd: 40 Last Error: No such device
 26/09 15:38:37,304 ERROR [140622254040832] (backend-v4l2.cpp:1235) xioctl(VIDIOC_DQBUF) failed for fd: 37 Last Error: No such device
 26/09 15:38:37,304 ERROR [140622126360320] (backend-v4l2.cpp:1235) xioctl(VIDIOC_DQBUF) failed for fd: 40 Last Error: No such device
[ WARN] [1569505117.305060150]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for fd: 40 Last Error: No such device,1.56951e+12,Error,Unknown Error
[ WARN] [1569505117.305297789]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for fd: 37 Last Error: No such device,1.56951e+12,Error,Unknown Error
 26/09 15:38:37,545 ERROR [140622245648128] (types.h:304) Cannot open '/dev/video0 Last Error: No such file or directory
 26/09 15:38:37,545 ERROR [140622245648128] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: Cannot open '/dev/video0 Last Error: No such file or directory
[ERROR] [1569505117.848122820]: The device has been disconnected!
 26/09 15:38:38,038 ERROR [140622480664320] (types.h:304) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
[ INFO] [1569505118.041744904]: Checking new devices...
[ INFO] [1569505118.051342241]: getParameters...
[ INFO] [1569505118.374118949]: setupDevice...
[ INFO] [1569505118.374209425]: JSON file is not provided
```

nacorti, 2019-10-18:
Were you ever able to resolve this issue? I'm going through the same thing (except the camera runs for maybe 2 min and then quits) with the same cryptic `Bond broken, exiting` message in the logs.

I have a hunch that my USB cord needs to be replaced, but until I can get a new one in the mail I'd like to eliminate other possibilities

VorpalBlade, 2019-10-18:
No, in the end I switched away from using Realsense cameras to a different brand with better range and less noise (that also uses OpenNI instead of it's own custom software stack, and that don't need a patched kernel module). However, before that I had a workaround in the form of a script that just relaunched the roslaunch file should it exit unexpectedly:

```bash
while true; do
	roslaunch my_package camera_realsense.launch
	# Give everything some time to settle after a crash
	sleep 1
done
```

Still, if it happens every 2 minutes, that seems far worse, and maybe even a different issue altogether.

nacorti, 2019-10-18:
I think it may also have something to do with overheating: when I run the camera in a darker environment (where it doesn't have to run the emitter with as much power) it seems to stay on indefinitely. Would need to test in bright environment with lots of cooling to see if that's true, but I've noticed the ASIC temps getting into the mid-40s easily.

What camera did you end up switching to?

VorpalBlade, 2019-10-18:
Orebec Astra Pro. It has been working okay for my use case. It has lower resolution images, but lower noise in the depth data (thanks to being a structured light camera as opposed to a stereo camera).

I was running the Intel RealSense camera in a room with AC (also far from the equator, so it is not very hot anyway). But yes it did feel hot to the touch after running for a while.

RealSenseCustomerSupport, 2019-11-20:
----------------------------------------------

Hi @VorpalBlade,

Although you switch to another vendor's camera, can I know what the versions of D435 firmware, ROS wrapper and librealsense you use to hit this issue in a long-run test? Can the D435 be still alive to have the streaming?
Note: The thermal condition needs to be concerned based on each component limits.

VorpalBlade, 2019-11-20:
I no longer have access to the physical camera device (having given it back to researcher I borrowed it from), nor did I keep the packages installed after switching camera. However this might give some clues as to the versions I could have used:
* I updated the firmware to the latest available in mid September.
* I was running the latest librealsense available via [these instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) as of the date of reporting the bug.
* The ROS wrapper was installed via apt. So it either came from ROS Kinetic or from the apt repository in the link in the previous bullet point, again the latest version as of writing the bug report.

I'm sorry but I do not understand what you mean by the question "Can the D435 be still alive to have the streaming?"

As for thermal considerations, the camera was mounted about 10 cm above the robot frame on a ball joint mount using the camera screw mount at the bottom, so it should have had plenty of ventilation. It was used indoors with AC, plus it was used in Sweden, so it wasn't hot anyway.

RealSenseCustomerSupport, 2019-11-21:
----------------------------------------------

Thanks for the info @VorpalBlade. We will continue improving our product's quality and reliability per the use cases.

closed 2019-11-21 by RealSenseCustomerSupport

bibbygoodwin, 2020-10-08:
**Crash _after a few seconds_!**
We are likewise getting a `Bond broken, exiting` message in an otherwise rather uninformative error log. Our camera crashes a few seconds after a "successful" `roslaunch` (at the end of which we see `RealSense Node Is Up!`), with a `process has died ` message (full snippet below).

Has this been resolved  since this issue was posted?

Many thanks!

```
[camera/realsense2_camera_manager-1] process has died [pid 10128, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/jattias/.ros/log/9f4325a6-0965-11eb-810d-e454e85fcaa1/camera-realsense2_camera_manager-1.log].
log file: /home/jattias/.ros/log/9f4325a6-0965-11eb-810d-e454e85fcaa1/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/jattias/.ros/log/9f4325a6-0965-11eb-810d-e454e85fcaa1/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```