realsenseai/realsense-ros #935: Update to newest librealsense

kamilritz, 2019-09-27:
When can the updated librealsense version be expected to go in the ROS wrapper?

doronhi, 2019-10-06:
You can try building with the latest librealsense version. If no interface modifications were done it should work.

kamilritz, 2019-10-14:
Thanks for you reply. I tried, but while launching the `rs_t265.launch`, I got this error:
```
[ERROR] [1571060191.657227981]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: API version mismatch: librealsense.so was compiled with API version 2.29.0 but the application was compiled with 2.25.0! Make sure correct version of the library is installed (make install)
```
Is there something I can do about this?

kamilritz, 2019-10-14:
I had to perform a `catkin clean` in order to build the ROS wrapper with the right librealsense version.

closed 2019-10-14 by kamilritz