realsenseai/realsense-ros #937: Setting frame queue size

Brudasoe, 2019-09-30:
Hey, I want to follow up [this issue](https://github.com/IntelRealSense/librealsense/issues/4955#issue-500247500) on the main realsense repository. My question for now is, is there an easy way to set the frame queue capacity via the ROS interface?

doronhi, 2019-10-06:
As opposed to the latency tool, realsense2_camera does not use pipeline but callbacks where getting frames from librealsense. Therefor there is no "queue capacity" to control.

Brudasoe, 2019-10-06:
So this means that the latency isn't tied to the fps?

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@Brudasoe Seen from the testing result, the latency by latency-tool below should be something related to the fps.
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/latency-tool

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@Brudasoe Any other questions about this ticket? Looking forward to your update. Thanks!

Brudasoe, 2020-02-13:
Thanks for following up on that, I did some tests, and it seems that latency did improve with more fps, even though I didn't check that thoroughly.

 I had a problem that I had calibrated the position of the camera with respect to the robot arm (hand-eye calibration), and after that I wanted to do a scanning motion, and it seemed like the camera was a little lagging with what I had expected, and that caused some issues, more with slower fps. 
I guess a way to solve that would be to synchronize internal Realsense clock with other clocks (for example with one on robot). But it's a little hard to do that programmatically without modifying the camera's internal code.

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@Brudasoe Please check below document to see if you can apply HW sync between the camera and  the robot.
https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

doronhi, 2020-02-16:
The timestamp marked with "Global Timestamp domain" is synchronized to the host clock.
You could synchronize the host clock with the robot's clock (or if they are the same, the problem is solved).

RealSenseCustomerSupport, 2020-02-27:
----------------------------------------------

@Brudasoe Any other questions about this? Thanks!

Brudasoe, 2020-03-04:
No, thanks, I am good for now :slightly_smiling_face:

RealSenseCustomerSupport, 2020-03-09:
----------------------------------------------

@Brudasoe Glad to hear that. Close this accordingly. Thanks!

closed 2020-03-09 by RealSenseCustomerSupport