realsenseai/realsense-ros #938: No D415 Sensors recognised on ROSLaunch

GnomeZone, 2019-09-30:
Hi, 
I've followed the instructions here to install the relevant packages to use a realsense D415 through ROS, but when launching the realsense2_camera launch file, none of the sensors in the camera are recognised. The terminal reads as follows:

started roslaunch server http://Sir-Clive:38175/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6019]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2a003f52-e393-11e9-a39c-94b86d5a68c4
process[rosout-1]: started with pid [6032]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [6049]
process[camera/realsense2_camera-3]: started with pid [6050]
[ INFO] [1569855682.296742187]: Initializing nodelet with 8 worker threads.
[ INFO] [1569855682.359808724]: RealSense ROS v2.2.8
[ INFO] [1569855682.359840818]: Running with LibRealSense v2.29.0
[ INFO] [1569855682.998306328]: getParameters...
[ INFO] [1569855683.137558772]: setupDevice...
[ INFO] [1569855683.137585972]: JSON file is not provided
[ INFO] [1569855683.137611841]: ROS Node Namespace: camera
[ INFO] [1569855683.137637112]: Device Name: Intel RealSense D415
[ INFO] [1569855683.137647774]: Device Serial No: 851112061486
[ INFO] [1569855683.137658877]: Device FW version: 05.11.11.100
[ INFO] [1569855683.137671544]: Device Product ID: 0x0AD3
[ INFO] [1569855683.137684631]: Enable PointCloud: Off
[ INFO] [1569855683.137693596]: Align Depth: Off
[ INFO] [1569855683.137712456]: Sync Mode: Off
[ INFO] [1569855683.137803727]: Device Sensors: 
[ INFO] [1569855683.137840783]: Stereo Module was found.
[ INFO] [1569855683.137862354]: RGB Camera was found.
[ INFO] [1569855683.137894145]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1569855683.137909816]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1569855683.137922278]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1569855683.137934779]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1569855683.137947592]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1569855683.137959840]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1569855683.137984753]: num_filters: 0
[ INFO] [1569855683.137998100]: Setting Dynamic reconfig parameters.
[ INFO] [1569855683.294383464]: Done Setting Dynamic reconfig parameters.
[ INFO] [1569855683.322989770]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1569855683.323435375]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1569855683.323830859]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1569855683.335890677]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1569855683.335934236]: setupPublishers...
[ INFO] [1569855683.339012062]: Expected frequency for depth = 30.00000
[ INFO] [1569855683.367862512]: Expected frequency for infra1 = 30.00000
[ INFO] [1569855683.393265563]: Expected frequency for infra2 = 30.00000
[ INFO] [1569855683.417269914]: Expected frequency for color = 30.00000
[ INFO] [1569855683.444677855]: setupStreams...
[ INFO] [1569855683.445791759]: insert Depth to Stereo Module
[ INFO] [1569855683.445838007]: insert Color to RGB Camera
[ INFO] [1569855683.445864730]: insert Infrared to Stereo Module
[ INFO] [1569855683.445900017]: insert Infrared to Stereo Module
[ INFO] [1569855683.466494544]: SELECTED BASE:Depth, 0
[ INFO] [1569855683.515947053]: RealSense Node Is Up!

As can be seen, both Fisheye cameras return the message "sensor isn't supported by current device".

When using real sense viewer, both cameras work as expected. For reference, uname returns the following: 
Linux Sir-Clive 4.15.0-64-generic #73~16.04.1-Ubuntu SMP Fri Sep 13 09:56:18 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

FrederikHaa, 2019-10-01:
There is no problem. The D415 does not have Fisheye sensors, and thus the module is skipped.

GnomeZone, 2019-10-01:
My mistake, thank you for your help. I was attempting to add the images as cameras instead, hence the feed displaying as black.

closed 2019-10-01 by GnomeZone