realsenseai/realsense-ros #939: 'Cannot open /dev/video10' when using 3 D435 with rs_multiple_devices.launch

FrederikHaa, 2019-10-01:
# System
OS: Ubuntu 16.04/18.04
Kernel: 4.15/5.0
librealsense: 2.29.0
realsense-ros: 2.2.8

# Problem
I'm trying to launch the multiple camera node with 3 D435 camera on my desktop running 18.04 and a NUC running 16.04.
I have filled in the serial number for all the cameras and changed the names, but otherwise not altered the launch file.

When I launch it on either device, the two first cameras launches fine (all topics are published correctly), but the third does not.

# Error
I get the following errors:
```
01/10 12:34:36,998 WARNING [140577662723840] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1569926076.999064102]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 01/10 12:34:37,000 WARNING [140369955493632] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1569926077.001299462]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 01/10 12:34:37,186 ERROR [140369865996032] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error
 01/10 12:34:37,187 ERROR [140369955493632] (types.h:304) Cannot open '/dev/video10 Last Error: No such file or directory
 01/10 12:34:37,187 ERROR [140370126038784] (types.h:304) Cannot open '/dev/video10 Last Error: No such file or directory
[ERROR] [1569926077.187108477]: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error
 01/10 12:34:37,187 ERROR [140370126038784] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: Cannot open '/dev/video10 Last Error: No such file or directory
 01/10 12:34:37,187 ERROR [140369955493632] (synthetic-stream.cpp:44) Exception was thrown during user processing callback!
[ERROR] [1569926077.187279825]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
 01/10 12:34:37,187 ERROR [140369955493632] (backend-v4l2.cpp:276) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 32 error: No such device
 01/10 12:34:37,187 ERROR [140369955493632] (types.h:304) xioctl(VIDIOC_DQBUF) failed for metadata fd: 35 Last Error: No such device
 01/10 12:34:37,187 ERROR [140369955493632] (backend-v4l2.cpp:1235) xioctl(VIDIOC_DQBUF) failed for metadata fd: 35 Last Error: No such device
[ WARN] [1569926077.187565013]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for metadata fd: 35 Last Error: No such device,1.56993e+12,Error,Unknown Error

```

The full cmd log can be seen at [pastebin](https://pastebin.com/1gS2rYXd)

If i remove one of the cameras from the launch file, everything works fine.

RealSenseCustomerSupport, 2020-01-23:
----------------------------------------------

@FrederikHaa,

Do you still need help with this question?

FrederikHaa, 2020-01-26:
We ended up using two computers with the cameras split between them, so I'll need to check again to see if the problem is still there with the newer versions of realsense.

I'll check next week.

RealSenseCustomerSupport, 2020-01-28:
----------------------------------------------

@FrederikHaa,

Thank you for the update. Please let us know the results of your check and whether you need any further assistance with this matter.

FrederikHaa, 2020-02-04:
It works now.

The error must've been fixed by a later version of realsense.

closed 2020-02-04 by FrederikHaa