realsenseai/realsense-ros #940: T265 Localization in existing map

tim-fan, 2019-10-02:
Hi there, I'm trying to use the T265 as a localization sensor for a robot project, but can't figure out how.

I'd like to map an area with the camera, save that map, then in future have the T265 report it's location in that map's coordinate frame. A simple use case being the ability to send the robot to certain pre-defined locations.

Is this achievable with the current sdk+ros-driver?

The closest I could find was [this issue](https://github.com/IntelRealSense/librealsense/pull/3324), which places a static-node at 0,0,0 during mapping, then tracks the position of that node when localising to determine location in the original map. However that method seems like a somewhat indirect approach to getting coordinates in the map frame, and also doesn't seem to be supported by realsense-ros, which does not report static node locations, as far as I can tell.

Any suggestions for paths forward?

Regards,
-Tim

TobiMiller, 2019-10-29:
you get an odometry from the t265. for mapping you have to use a laserscanner or a stereo-camera system.

tim-fan, 2020-01-14:
Thanks for the response, yes I have seen that method for integrating the T265 into a localisation system. 

What I was hoping for was to have the T265 perform localisation independently, without the need to integrate/fuse with other sensors. 

When I bought the device I understood that this functionality would be supported (as implied by, for instance, [this comment](https://github.com/IntelRealSense/librealsense/issues/3129#issuecomment-456980049)). I also assumed that the functionality would be available for use through the ROS API. However as far as I can tell, this support is still not available, hence I opened this issue.

Michael-Equi, 2020-05-15:
If I am not mistaken, this requires updating the ros node with some of the functionality already provided with the librealsense2 library. This is a feature that is absolutely required for the project I am working on so if nobody else is working on implementing it I will give it a shot. Has anyone made progress on adding this feature to the T265 ROS node that I should be aware of before I start?

tim-fan, 2020-05-15:
Hello, personally I haven't looked into this since I logged the issue, so I'm not aware if anyone else has made progress. That would be awesome if you could implement it :thumbsup:

arjunsengupta98, 2020-07-30:
Hello!
Has anyone been able to make any progress?
@Michael-Equi  @tim-fan ?

> When I bought the device I understood that this functionality would be supported (as implied by, for instance, this comment). I also assumed that the functionality would be available for use through [...]

@tim-fan I too assumed the same. 

Right now, I'm simply using the odometry readings from the t265 and depth information from D435 to create a 2D occupancy map on ROS. If I'm not wrong, that's what they have done in the occupancy branch of realsense-ros?https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping 
Isn't his simply using the T265 like an IMU? We aren't using any online SLAM capabilities of the T265 here, are we?

@dorodnic  Your comments would be appreciated as well.

Michael-Equi, 2020-07-30:
@arjunsengupta98 I am currently working on a simplified T265 only node for ROS2 that includes relocalication and wheel odometry input. Its not all that refined yet but it does work. You can find my custom node here https://github.com/BytesRobotics/ros2_t260/tree/development

arjunsengupta98, 2020-08-03:
Hey @Michael-Equi 
Thanks for the link!
Any idea on how the internal map can be continuously streamed while being formed? (Something like how rviz depicts the occupancy map continuously while it is being formed)? 
Does your node do that? I haven't worked on ROS2 (I'm a novice in ROS itself), hence thought about confirming this before going ahead with an attempt to implement your node.

Michael-Equi, 2020-08-03:
@arjunsengupta98  I'm not sure why you would want the map streamed since its not really interpretable unless you can figure out how intel encodes it but I guess if you really want it frequently updated you could just call the save map service often and then read the file as needed. Really what map saving and loading is for is so that you can get the transform between a current realsense map/pose to a past one basically allowing you to align it with another map type like an occupancy map handled by ROS.

arjunsengupta98, 2020-08-10:
Sure, that makes perfect sense. Turns out that I had slightly misunderstood the concept.
Thanks a ton!

> @arjunsengupta98 I'm not sure why you would want the map streamed since its not really interpretable unless you can figure out how intel encodes it but I guess if you really want it frequently [...]

Sure, that makes perfect sense. Turns out that I had misunderstood the concept.
Thanks a ton!

closed 2021-03-24 by RealSenseSupport

tim-fan, 2021-03-24:
@RealSenseSupport is this closed because localisation in an existing map will not be supported? Or has this been implemented?