realsenseai/realsense-ros #941: AprilTag support with Realsense-ros wrapper

jacksparkman, 2019-10-03:
Will the example codes be added to the realsense-ros wrapper? The example code rs-pose-apriltag is great, but cannot be published to a ROS topic. Being able to detect apriltags with the realsense cameras inside ROS would be great for development.

stasgurevich, 2019-10-14:
Here is the ROS package that can be used to detect apriltags with any cam (including Realsense).
https://github.com/AprilRobotics/apriltag_ros

RealSenseCustomerSupport, 2019-12-12:
----------------------------------------------

@jacksparkman Did you get it through? Any other questions? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@jacksparkma Didn't get response from you for weeks. Will close the ticket at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-02-03 by RealSenseCustomerSupport

AndreV84, 2021-10-04:
how exactly to run it with realsense d435? Thanks