realsenseai/realsense-ros #943: Colorize point cloud based on 415 left IR (color) camera

mpowelson, 2019-10-04:
The 415 left IR camera can capture color images which should be able to be applied to the point cloud. This is recommended in [THIS](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf) document since they are already pixel aligned and time synced. 

I can do this on a 435 by using the launch file below 
```
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
      <arg name="serial_no"         value="$(arg serial_no)"/>
      <arg name="json_file_path"    value="$(arg json_file_path)"/>
      <arg name="enable_color"      value="false"/>
      <arg name="filters"           value="pointcloud"/>
      <arg name="pointcloud_texture_stream" value="RS2_STREAM_INFRARED" />
      <arg name="pointcloud_texture_index" value="1" />
      <arg name="tf_prefix"         value="$(arg camera)"/>
    </include>
```

This appears to return a pointcloud with the intensity colored from the IR image. However, running the same command on the 415 seems to do the same. It seems that RS2_STREAM_INFRARED always implies single channel, but changing those fields did not seem to help. What should I change in order to enable color on the left camera of the 415 and colorize the point cloud with it?

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@mpowelson Could you please try with the latest SDK and wrapper version? We couldn't reproduce the issue with the latest version. Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@mpowelson Any update from your side? Thanks!

drchrislewis, 2019-12-13:
I switched to using the 435 for my application. I also switched to using depth_image_proc instead using the onboard one since I needed a structured pointcloud. I do plan to switch back and test this again when I get time, but that could be a few weeks. You can close this in the meantime if you want.

mpowelson, 2019-12-13:
I switched to using the 435 for my application. I also switched to using depth_image_proc instead using the onboard one since I needed a structured pointcloud. I do plan to switch back and test this again when I get time, but that could be a few weeks. You can close this in the meantime if you want.

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

@drchrislewis Thanks for the update! Please feel free to create another one if you still have questions or issues later.

closed 2019-12-17 by RealSenseCustomerSupport