realsenseai/realsense-ros #944: How can I get the gray scale image instead of rgb one?

studennis911988, 2019-10-04:
Hi, since I'm using realsense d435i for my SLAM project, but I notice that when I try to stream the rgb image of resolution 640x480 at 60fps and depth image of the same parameters, and I can only get 30fps from the rgb image. 
I think it is due to overwhelming transmitting over the usb3.0, so I wonder which file in the SDK did the ros_wrapper(rs_camera.launch) called?
 I wish I could modified the original code that allows me stream the gray image instead of the rgb one, in order to get higher fps. Wish you can give me some hint.
Thank you!

mpowelson, 2019-10-04:
There are other ways I'm sure, but this is one way to get grayscale

```    
<include file="$(find realsense2_camera)/launch/rs_camera.launch">
      <arg name="enable_color"      value="false"/>
      <arg name="filters"           value="pointcloud"/>
      <arg name="pointcloud_texture_stream" value="RS2_STREAM_INFRARED" />
      <arg name="pointcloud_texture_index" value="1" />
</include>
```

doronhi, 2019-10-06:
realsense2_camera does not have an interface to publish the rgb image as a grayscale image.

I'm not sure what hardware you are using but it is most likely that the problem with 60 fps is not related to it but to the auto exposure settings of the camera. By default, for the RGB camera, auto exposure has higher priority that the frame rate required. If you want to override this behavior add the following lines to your launch file:
```
  <rosparam>
      /camera/rgb_camera/auto_exposure_priority: false
  </rosparam>
```
Or just test toggling on and off the `auto_exposure_priority` flag using `rqt_reconfigure.`

studennis911988, 2019-10-06:
@mpowelson @doronhi  Thanks for your reply! Really helpful!

closed 2019-10-06 by studennis911988