realsenseai/realsense-ros #947: Pausing the camera

myleslidka, 2019-10-11:
Does anybody know a way to pause the camera? I working on an application that is using 3 d435s connected to a jetson nano and only need them running for short periods of time.

I have tried just killing the camera nodes but that gives me problems when trying to restart them. I've tried using roslaunch.shutdown to stop the cameras nodes (which I am starting from a launch file similar to rs_multiple_devices.launch) but then when I try to start the launch file again I get" bad roslaunch process monitor initialization: process monitor is already dead".

If anyone has any suggestions it would be greatly appreciated.

RealSenseCustomerSupport, 2019-11-19:
----------------------------------------------

Hi @myleslidka,

Please check below to see if it can be used for your use case.
https://github.com/IntelRealSense/realsense-ros/pull/487

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

I will close this if no further question.

closed 2019-12-06 by RealSenseCustomerSupport

chia-labradorsystems, 2020-04-06:
Hi,  @RealSenseCustomerSupport 
Seems like the solution provided in #487 is no longer available now.
`void BaseRealSenseNode::toggleSensors(bool enabled) ` is still there but there is no dynamic parameter hooks with it right now.

doronhi, 2020-04-07:
You are right. I'll try to attend to it as soon as I manage.

chia-labradorsystems, 2020-04-07:
@doronhi Thank you.
I tried to implement it myself yesterday, `toggleSensors` can stop the camera but cannot start it again
I'm using d435 with latest firmware.