realsenseai/realsense-ros #948: crash

comwise, 2019-10-12:
Required Info |  
-- | --
Camera Model | D435i
Firmware Version |  
Operating System & Version | Ubuntu 16.04
Kernel Version (Linux Only) | 4.15.0-15-generic
Platform | intel PC
SDK Version | 2.25
Language | C++/python
Segment | Robot

the ros node will be crashed when the node is running sometime
you can view the crash bug, if you need detail info, i will offer you later

i want to know who asked this problem?

i look forwark  to your reply 

== SourcePackage =================================
ros-kinetic-nodelet

== Stacktrace =================================
#0  0x00007f7773501418 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
        resultvar = 0
        pid = 8024
        selftid = 8077
#1  0x00007f777350301a in __GI_abort () at abort.c:89
        save_stage = 2
        act = {__sigaction_handler = {sa_handler = 0x0, sa_sigaction = 0x0}, sa_mask = {__val = {0, 0, 140151012739887, 140151016195392, 65, 0, 0, 0, 140151022051184, 140150747162256, 140150747162256, 1, 35429216, 140150554237904, 66, 140150747162368}}, sa_flags = 1941141632, sa_restorer = 0x7f7758001cb8}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007f7773b3a84d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#3  0x00007f7773b386b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#4  0x00007f7773b376a9 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#5  0x00007f7773b38005 in __gxx_personality_v0 () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#6  0x00007f77738a4f83 in ?? () from /lib/x86_64-linux-gnu/libgcc_s.so.1
No symbol table info available.
#7  0x00007f77738a5487 in _Unwind_Resume () from /lib/x86_64-linux-gnu/libgcc_s.so.1
No symbol table info available.
#8  0x00007f77682cb8c6 in realsense2_camera::RealSenseNodeFactory::closeDevice() () from /opt/jz/jrosth/lib//librealsense2_camera.so
No symbol table info available.
#9  0x00007f77682cb8f8 in realsense2_camera::RealSenseNodeFactory::~RealSenseNodeFactory() () from /opt/jz/jrosth/lib//librealsense2_camera.so
No symbol table info available.
#10 0x00007f77682cb98e in realsense2_camera::RealSenseNodeFactory::~RealSenseNodeFactory() () from /opt/jz/jrosth/lib//librealsense2_camera.so
No symbol table info available.
#11 0x00007f77750b6d3f in void class_loader::ClassLoader::onPluginDeletion<nodelet::Nodelet>(nodelet::Nodelet*) () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#12 0x00007f77750af6fa in ?? () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#13 0x00007f77750b463d in nodelet::ManagedNodelet::~ManagedNodelet() () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#14 0x00007f77750b0f8d in nodelet::Loader::unload(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#15 0x00007f77750bc15b in nodelet::LoaderROS::unload(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/kinetic/lib/libnodeletlib.so
No symbol table info available.
#16 0x00007f7774c67141 in bond::Bond::flushPendingCallbacks() () from /opt/ros/kinetic/lib/libbondcpp.so
No symbol table info available.
#17 0x00007f7774c678a5 in bond::Bond::bondStatusCB(boost::shared_ptr<bond::Status_<std::allocator<void> > const> const&) () from /opt/ros/kinetic/lib/libbondcpp.so
No symbol table info available.
#18 0x00007f7774c6e31c in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<bond::Status_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<bond::Status_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<bond::Status_<std::allocator<void> > const>) () from /opt/ros/kinetic/lib/libbondcpp.so
No symbol table info available.
#19 0x00007f7774c7106c in ros::SubscriptionCallbackHelperT<boost::shared_ptr<bond::Status_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) () from /opt/ros/kinetic/lib/libbondcpp.so
No symbol table info available.
#20 0x00007f77749f6d8d in ros::SubscriptionQueue::call() () from /opt/ros/kinetic/lib/libroscpp.so
No symbol table info available.
#21 0x00007f777499c838 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/kinetic/lib/libroscpp.so
No symbol table info available.
#22 0x00007f777499e23b in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/kinetic/lib/libroscpp.so
No symbol table info available.
#23 0x00007f77749fa284 in ros::AsyncSpinnerImpl::threadFunc() () from /opt/ros/kinetic/lib/libroscpp.so
No symbol table info available.
#24 0x00007f7772a565d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
No symbol table info available.
#25 0x00007f777282f6fa in start_thread (arg=0x7f77637fe700) at pthread_create.c:333
        __res = <optimized out>
        pd = 0x7f77637fe700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140150747162368, -2386450790782055189, 0, 140730369680367, 140150747163072, 35546984, 2310163376555969771, 2310199977450032363}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>

RealSenseCustomerSupport, 2019-12-12:
----------------------------------------------

@comwise Could you please tell which launch file do you use when get the issue? And did you get chance to try the latest SDK and ROS version to see if still have issue? Thanks!

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@comwise Do you still have this issue? Could you please update? Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@comwise Sorry that we didn't get response from you for weeks. Will close this issue at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-02-03 by RealSenseCustomerSupport