realsenseai/realsense-ros #949: No infra2 node

cugcsu, 2019-10-14:
I run this command：
`roslaunch realsense2_camera rs_camera.launch` 
and
`rostopic list`
/camera/accel/imu_info
/camera/accel/sample
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/gyro/imu_info
/camera/gyro/sample
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra1/image_rect_raw/theora
/camera/infra1/image_rect_raw/theora/parameter_descriptions
/camera/infra1/image_rect_raw/theora/parameter_updates
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/pointcloud/parameter_descriptions
/camera/pointcloud/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf_static

I found that there are no infra2 node in these nodes. At the same time, I open rviz, there is no such node, and no depth image is displayed.
But I can see the depth image with realsense-viewer.

doronhi, 2019-10-15:
I'm not really sure I understand the issue:
I see you have the `/camera/depth/image_rect_raw` which is the depth image. So, if you do `rostopic hz /camera/depth/image_rect_raw` and see a reasonable rate, the problem of presenting it in RViz is another issue.
I also see the topic `/camera/depth/color/points` which suggests you ran the node using the flag "filters:=pointcloud" or the old format "enable_pointcloud:=true". Could you have changed also something else, either in the command or in the launch file, to prevent the topic "/camera/**infra2**/image_rect_raw" from being published?
If not, could you post the running log of realsense2_camera?
In any case, the fact that the topic "/camera/infra2/image_rect_raw" is not published does not prevent the creation and publication of the depth topic.

RealSenseCustomerSupport, 2019-11-25:
----------------------------------------------

@cugcsu Could you please post the running log so that we can look into it further? Thanks!

closed 2019-12-10 by cugcsu