realsenseai/realsense-ros #950: API version mismatch: librealsense.so was compiled with API version 2.28.1 but the application was compiled with 2.25.1!

studennis911988, 2019-10-14:
Hi, I install **realsense SDK (version 2.28.1)** for D435i on **NVIDIA TX2** and connect to D435i **successfully with realsense-viewer**.
[image]
But, when I run the [ROS Wrapper](https://github.com/IntelRealSense/realsense-ros), I counter the error : "API version mismatch: librealsense.so was compiled with API version 2.28.1 but the application was compiled with 2.25.1! " . I know that it is due to the mismatching between realsense library and the API version of the ros-wrapper used, but when I install the SDK back to version 2.25.0, I can't even get the realsense-viewer connect to the camera.
My question is that is there **another solution for this while keeping the SDK version(2.28.1) ?**
Thanks in advance!

FrederikHaa, 2019-10-25:
You need to clean and rebuild the realsense ros package

oka1125, 2019-11-21:
@studennis911988 

I have met the same problem.
I found ros-kinetic-librealsense-camera package was installed other than librealsense2 which I manually compiled and installed.
Check by this command:
  dpkg -l | grep librealsense

I soluted this problem by following command:
  sudo apt remove ros-kinetoc-librealsense2

RealSenseCustomerSupport, 2020-01-02:
----------------------------------------------

Hi Dennis,

Are you still having this issue? Can this be closed? Another option is to add a "copy if newer" step to your pre-build process to guarantee the object file versions always match.

studennis911988, 2020-01-03:
@RealSenseCustomerSupport  Sorry for late response. Problem solved!

closed 2020-01-03 by studennis911988