realsenseai/realsense-ros #951: Issue with realsense imu

aswinkumar2019, 2019-10-14:
Hello guys,
                 We tried to implement a project where( https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i) slam is part of it.The data from imu appears to be slow,the camera_link frame appears to respond very slow and sometimes,wrong.
It is clearly mentioned on that website as,

****The built-in IMU can only keep track for a very short time. Moving or turning too quickly will break the sequence of successful point cloud matches and will result in the system losing track. It could happen that the system will recover immediately if stopped moving but if not, the longer the time passed since the break, the farther away it will drift from the correct position. The odds for recovery get very slim, very quickly. The parameters set in the launch file are most likely not ideal but this is a good starting point for calibrating****

Is the response from imu very slow to be relied or,is there any way to improve the data from imu????

doronhi, 2019-10-15:
The data from the IMU comes at a high rate as you can see from the spec and the response is quick as you can see by checking the raw messages coming out of realsense2_camera. 
I suspect the integration with RTabMap in that demo is what causing the visible slowness. I feel that with more time and better understanding of RTabMap's parameters and requirements the demo could run much smother.

aswinkumar2019, 2019-10-15:
I didnt modify anything in that launch file.
The imu_topic subscribes to /imu/data topic by default so,i thought there is no need for modifying anything related to imu.
Is there anything i could modify to find the issue???

doronhi, 2019-10-15:
I think I wasn't clear enough.
I believe the original launch file in the demo is not good enough and does not pass the IMU information well enough to the RTabMap SLAM engine. I believe that it can be much improved with better knowledge of RTabMap.

araujokth, 2019-11-08:
Hi @aswinkumar2019 , we also faced the same problem when using the D435i with RTABmap. The problem is that RTABmap expects the IMU measurements to be periodic at the set rate, which is not the case with the current ros driver as its discussed here https://github.com/IntelRealSense/realsense-ros/issues/898. This would have to be fixed before you can use it.

closed 2019-11-30 by aswinkumar2019

doronhi, 2019-12-04:
Because these issues are very similar, please check my comment [here](https://github.com/IntelRealSense/realsense-ros/issues/898#issuecomment-561675431)