realsenseai/realsense-ros #955: T265 gives out NaNs

manomitbal, 2019-10-17:
Hi!

I have been using the output of /camera/odom/sample within my SLAM package and I have been consistently facing this issue : whenever there is a halt in robot motion after continuous usage, the output of the above mentioned topic turns to NaNs and my SLAM node crashes, but the realsense node does not, it continues to spit out NaNs. This happens when you stop moving after a while even in a feature rich environment. I am not sure whether this has something to do with how the transform tree is set up. This issue is usually resolved by me unplugging the T265 from the USB, plugging it again and starting the node. Sometimes it goes away with some additional motion. Is this happening to other people as well?

imasoul2, 2019-10-19:
@manomitbal , I'm experiencing the same issue, my project is about using the T265 for navigation and this occurs almost 2 out of 5 times.

BriceRenaudeau, 2019-10-21:
I had the same problem lately, I made a node to detect this (and the diverging values) in order to kill and restart the node of the camera. I hope it will be fixed soon.

reyes73a, 2019-10-21:
I have also seen this several times and its quite annoying and worrisome because the T265 is a device that should be close to 100% reliable or else is useless for practical applications. If a robot depends on this and it happens it will have a dramatic failure. I usually see it when leaving the node run for several hours. However, it is not 100% consistent making me think that there is something else triggering this. I only see it when using ROS, not when using C++ directly.

manomitbal, 2019-10-24:
Just an update, I tried to set RS2_OPTION_ENABLE_POSE_JUMPING and RS2_OPTION_ENABLE_RELOCALIZATION as false thinking that maybe this might be causing the NaN issue, but it still occurs

RealSenseCustomerSupport, 2019-11-13:
----------------------------------------------

Hi @manomitbal

An updated LibRealSense release has been posted:
https://github.com/IntelRealSense/librealsense/releases/tag/v2.30.0

There are updates which can have a positive affect on the NaN issues that you are seeing.

Please give this new release a test in your environments and let us know what you are seeing.

Thanks

dj-missionsystems, 2019-11-20:
We are running a T265 on a quadcopter Into an NVIDIA Tx2 and are still seeing this problem with v2.30.0 firmware. Not very repeatable, so doubly frustrating (and not easy to power-cycle mid-flight). Are there any settings we should be playing with further that might help?

Cheers

RealSenseCustomerSupport, 2019-11-20:
----------------------------------------------

Hi @dj-missionsystems

Thanks for the response.  When you say not very repeatable, is there something that is constant about the flight or the environment or wind, etc that might be the same between the times you see drift and NaN and the times you don' see it?  Prior to the v2.30.0 was it very repeatable and you're saying now it's not as repeatable with this new version?  Unfortunately right now there are no settings that changes the behavior.

dj-missionsystems, 2019-11-21:
Unfortunately, during a 5-day flight-trial, weather meant we could only fly one afternoon. Prior to that we thought we had fixed the nan issue (which seemed to appear periodically after a few minutes) by turning USB autosuspend off (as per https://www.jetsonhacks.com/2015/05/27/usb-autosuspend-nvidia-jetson-tk1/). However we did two flights - one without NANs, and then one with. The only difference seemed to be the 2nd flight was in a more complex environment with more trees, and possibly more wind-turbulence. We thought it might be RF noise or voltage drop from the drone itself in flight, so tried to repeat things with the props upside-down on the ground but couldn't repeat the error. Nor with moving the aircraft around rapidly by hand. One remaining thought is that it might be vibration? I will try and do some more flight tests in the next few days and report back. (This is also using ROS, not C++.)

manomitbal, 2019-11-25:
@dj-missionsystems everytime now that I get NaNs are after an intense bout of vibration/jerks

RealSenseSupport, 2020-09-15:
Thank you for highlighting the unfortunate continuation of the NaN issue while using T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

cimbar, 2020-09-15:
> We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

Could you give an approximate date for next generation products?

RealSenseSupport, 2020-09-17:
Hi @cimbar 

For all publicly released products please monitor our site:  https://www.intelrealsense.com/

Any products under development that has not been released cannot be commented on.

closed 2020-09-23 by RealSenseSupport