realsenseai/realsense-ros #959: Realsense IMU Coordinate Frame

omotuyoa, 2019-10-22:
Hello,

I am using an intel realsense camera D435i to achieve visual-inertial slam based on the opensource_tracking.launch file. These were the following issues:

1. /imu/data is not publishing any data. I saw that the name of parameters passed into magwick filter were wrong e.g _use_ma is supposed to be use_mag as seen below:
https://github.com/IntelRealSense/realsense-ros/blob/2a45f09003c98a5bdf39ee89df032bdb9c9bcd2d/realsense2_camera/launch/opensource_tracking.launch#L11
I changed the name of parameters and /imu/data started publishing. However, the data published is using a coordinate frame that is different from what is on this page: https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/.

**What is the correct coordinate frame of the IMU with the camera facing forward?**

2. In this line, the world frame is set as **enu** but the z-data is for a forward/backward direction and the y-data is for a downward/upward direction. Is there any exchange of data between axis that needs to be done before passing it to magwick filter? **What is the correct world frame that should be used?**
https://github.com/IntelRealSense/realsense-ros/blob/2a45f09003c98a5bdf39ee89df032bdb9c9bcd2d/realsense2_camera/launch/opensource_tracking.launch#L13

**Kindly assist. Thank you**

aswinkumar2019, 2019-10-24:
I checked online regarding your imu question.Realsense uses bosch BMI055 imu
Just go through this document below page number 37.
(https://datasheet.octopart.com/BMI055-Bosch-datasheet-32868712.pdf)
The z axis is the axis pointing towards gravity.In realsense case it is y axis.
So there is change in coordinate system between sensor frame and world frame(realsense frame)
So,z axis changes to y.
So,i guess ENU is the correct coordinate system according to my knowledge(I maybe wrong)

omotuyoa, 2019-10-24:
Thank you @aswinkumar2019. If I am correct, ENU refers to the coordinate frame that maps to XYZ axes. First, with the camera facing forward on it stand, the value of y-axis linear acceleration is -9.8 which means the y-axis is pointing upward. This makes the data to be in the form of XZY. Will this not affect the estimate from the magwick filter? Is there any transformation to be done when passed into robot localization package since the coordinate frame is different from the one used by RTABMap? Thanks.

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@omotuyoa

Were you able to utilize the D435i in your usage model?  Is there anything else needed on this thread?

closed 2019-12-20 by RealSenseCustomerSupport