realsenseai/realsense-ros #961: T265 change contrast?

erikasnow, 2019-10-23:
Is there a set of parameters somewhere that I could use to change the brightness and contrast of one of the fisheye images? It looks like this sort of thing is accessible on other realsense cameras, but I can't find any parameters like that on the T265.

doronhi, 2019-10-27:
You can change the brightness and contrast of both fisheye cameras together through the tracking_module section in the dynamic_reconfigure params but you can't change each of them separately. I don't see such an option with other realsense devices also.
I think such an option could hamper with the image matching and the depth performance.
Why would you need such an option?

erikasnow, 2019-11-01:
I am trying to use the camera to detect apriltags with a drone. Seems like sometimes the camera has a hard time making detections because of shadows and other environmental factors. I was hoping that changing the contrast of the camera might mitigate some of these issues.

doronhi, 2019-11-10:
Sorry for my late response. 
So I guess you can change the exposure and gain controls for both cameras together and it will be ok?
Notice that you cannot change them while streaming so set the parameters in the launch file in a `<rosparam>` section.

RealSenseCustomerSupport, 2019-12-24:
----------------------------------------------

@erikasnow
Notice there hasn't been any response since doronhi provided some update here.  If there is nothing further, we can close out this thread.

closed 2020-01-15 by erikasnow

FergalLonerganKUL, 2020-02-04:
[ INFO] [1580847984.061964435]: Tracking Module was found.
[ INFO] [1580847984.062008884]: (Depth, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580847984.062032143]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580847984.062051935]: (Infrared, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580847984.062076800]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580847984.062100067]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580847984.062289429]: num_filters: 0
[ INFO] [1580847984.062309395]: Setting Dynamic reconfig parameters.
terminate called after throwing an instance of 'rs2::error'
  what():  To control exposure you must set sensor to manual exposure mode prior to streaming
[camera/realsense2_camera_manager-2] process has died [pid 17123, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/fergal/.ros/log/9c1a46f2-478c-11ea-9773-302432929cdf/camera-realsense2_camera_manager-2.log].
log file: /home/fergal/.ros/log/9c1a46f2-478c-11ea-9773-302432929cdf/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/fergal/.ros/log/9c1a46f2-478c-11ea-9773-302432929cdf/camera-realsense2_camera-3*.log

****************************************************************

Hi @doronhi I have followed the explanation you have givien above. My issue is is that I'm being told that the changes must be made before the camera begins to stream, however, I have added the rosparam section to the very beginning of my launch file. Any idea how to change it before the camera begins to stream? I also tried the "enable_pose_jumping" argument included which was in another issue thread, but this did not change the issue with the camera streaming before we have the chance to change the rosparams.

****************************************************
  
...
...
</launch>

FergalLonerganKUL, 2020-02-04:
My launch file looks like this:

<launch>
  <arg name="enable_pose_jumping" default="false"/>
  <rosparam>
    /camera/tracking_module/enable_auto_exposure: false
    /camera/tracking_module/exposure: 15000.0
    /camera/tracking_module/gain: 4.0
  </rosparam>
...
...
</launch>

FergalLonerganKUL, 2020-02-04:
```
 <launch>

  <arg name="enable_pose_jumping" default="false"/>
  <rosparam>
    /camera/tracking_module/enable_auto_exposure: false
    /camera/tracking_module/exposure: 15000.0
    /camera/tracking_module/gain: 4.0
  </rosparam>
```

FergalLonerganKUL, 2020-02-04:
My launch file looks like this:
```
<launch>

  <arg name="enable_pose_jumping" default="false"/>
  <rosparam>
    /camera/tracking_module/enable_auto_exposure: false
    /camera/tracking_module/exposure: 15000.0
    /camera/tracking_module/gain: 4.0
  </rosparam>
...
...
</launch>
```