realsenseai/realsense-ros #962: RealSense T265 Wheel Odometry configuration

victorbl, 2019-10-23:
I have integrated the T265 tracking camera into my robot stack and are seeing great results in general odometry tracking of the robot. I developed a small C++ package to stream the camera poses, and can create the appropriate transforms and topics. In an attempt to wrap up the package features, I have spent the last week trying to integrate the wheel odometry feature as expected by the T265 camera and have a few question to ensure I get this right.

For some brief details on my robot configuration, I am using the T265 on a differential drive robot, two motorized wheels centered on a circular base. I am using the Parallax Arlo robot platform. I have mounted the T265 on the upper deck, at the very front of the robot roughly .31 meters from the floor, and .2 meters forward from the center of the bot and wheels.

My questions

1. With the robot being differential drive (two motorized wheels), do I provide the T265 a single ‘Velocimeters’ block in the json file with the extrinsics transform between my robots ‘base_link’ and the ‘camera’. The odometry velocity I would send via ‘send_wheel_odometery’ would be the overall linear velocity of the robot from ‘base_link’ 

-or-

Do I provide the T265 two ‘Velocimeter’ definitions, each defining the extrinsics transform from each unique wheel? Then presumably the odometry velocity sent for each velocimeter definition would represent the unique linear velocity for each wheel.

2. Is there a way to verify the wheel odometry configuration and velocity stream provided to the T265 are behaving as expected? I tried covering the lenses of the camera, hoping it would fall back on wheel configurations and the data stream for localization, but saw no significant pose changes coming from the camera based on the wheel odometry. It’s possible I had this mis-configured and wanted to ask if there is a good way to verify the provided wheel configuration and velocity match the camera’s expectations and are oriented correctly with my robot’s transforms.

Thank you in advance for insights into these two questions. I’ve done some relatively deep digging around the internet and have had no luck getting clarity on these two open items in my implementation.

RealSenseCustomerSupport, 2019-12-09:
----------------------------------------------

Hi Victor,

1. Yes...the use of two velocimeters can give slightly better performance/accuracy because it has more parameters and can account for things like a difference in wheel diameters, etc but I don't believe the slight improvement in performance justifies the increased complexity.

2. If possible....you could lift the robot and just turn the wheels which should affect the estimated position. Even more if the camera is (partially) covered. (Not fully covered because it will switch to 3DOF, orientation only!)

I'd also recommend checking out the wheeled odometry python example here... https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/t265_wheel_odometry

As well as some updated documentation at the end of this md file...  https://github.com/IntelRealSense/librealsense/blob/development/doc/t265.md

RealSenseCustomerSupport, 2019-12-18:
----------------------------------------------

Does this answer your question Victor? If so we will close this ticket.

Thanks

victorbl, 2019-12-18:
Thank you, this was very helpful.  Appreciate you taking the time to reply!

closed 2019-12-18 by victorbl