realsenseai/realsense-ros #963: IMU output in imufiltermadgwick

aswinkumar2019, 2019-10-23:
I tried to run slam using realsense d435i using opensourcetracking.launch file.
The raw imu data from realsense appear to be good but, the output from imufiltermadgwick is not good.
The camera_link frame doesn't move most of the time and, even if it moves it is wrong.
The camera_link frame doesn't appear to rotate too. Please help me to fix this.
I am using jetson nano

aswinkumar2019, 2019-10-24:
Hey realsense team, found the issue.
Thanks to a guy named Oyindamola who had the same issue and helped me to fix this.The issue was with imufiltermadgwick,
the parameter _use_ma should be changed to use_mag.
The slam will not get proper imu data if this is not done.
I don't know why intel has not fixed this issue for a long time.
Please fix this soon so, others don't get into this issue

closed 2019-10-24 by aswinkumar2019

ashirsat, 2019-12-18:
Hi @aswinkumar2019 ,

Can you point me to the config file for the above modification?

Thanks.

aswinkumar2019, 2019-12-18:
Hello @ashirsat,the code has been modified already.
no changes need to be done now.
the realsense team has accepted the changes made.
The code should work fine now
Regards,
Aswin Kumar

ashirsat, 2020-03-27:
Hey @aswinkumar2019 ,

Thanks for the update I will pull the latest release and see if it is solved.

Regards,
Aniket