realsenseai/realsense-ros #968: D435 Crash

AutomatoRobotics, 2019-10-28:
Hello,

I'm using the D435 ROS node with SDK 2.2.8 on Ubuntu 16.04, and after a while (several minutes) my entire system crashes and shuts down.
I see several errors in the syslog log file:

```
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711119] Unknown pixelformat 0x00000000
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711129] WARNING: CPU: 4 PID: 13161 at /
build/linux-hwe-tUyiB5/linux-hwe-4.15.0/drivers/media/v4l2-core/v4l2-ioctl.c:129 3 v4l_fill_fmtdesc+0xf4d/0x1260 [videodev]
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711129] Modules linked in: cdc_acm hid_
generic hidp ccm rfcomm bbswitch(OE) uhid bnep snd_hda_codec_realtek snd_hda_cod
ec_generic snd_hda_codec_hdmi binfmt_misc nls_iso8859_1 arc4 intel_rapl x86_pkg_
temp_thermal intel_powerclamp coretemp nvidia_uvm(OE) iwlmvm kvm_intel mac80211 
hid_multitouch kvm irqbypass dell_laptop dell_wmi iwlwifi dell_smbios wmi_bmof d
cdbas crct10dif_pclmul crc32_pclmul ghash_clmulni_intel dell_wmi_descriptor mxm_
wmi dell_smm_hwmon cfg80211 pcbc snd_hda_intel aesni_intel snd_hda_codec aes_x86
_64 crypto_simd snd_hda_core glue_helper snd_hwdep cryptd intel_cstate intel_rap
l_perf snd_pcm snd_seq_midi snd_seq_midi_event input_leds snd_rawmidi joydev ser
io_raw uvcvideo snd_seq videobuf2_vmalloc snd_seq_device snd_timer videobuf2_mem
ops videobuf2_v4l2 videobuf2_core snd videodev
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711145]  soundcore media btusb btrtl bt
bcm btintel bluetooth ecdh_generic mei_me idma64 virt_dma mei shpchp intel_lpss_
pci processor_thermal_device intel_lpss intel_soc_dts_iosf int3403_thermal intel
_pch_thermal intel_hid int3400_thermal int3402_thermal acpi_pad wmi sparse_keyma
p int340x_thermal_zone acpi_thermal_rel mac_hid ppdev lp parport autofs4 nvidia_
drm(POE) nvidia_modeset(POE) nvidia(POE) i915 i2c_algo_bit drm_kms_helper syscop
yarea sysfillrect sysimgblt r8169 fb_sys_fops ahci ipmi_devintf psmouse mii drm 
libahci ipmi_msghandler i2c_hid hid video [last unloaded: parport_pc]
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711158] CPU: 4 PID: 13161 Comm: nodelet
 Tainted: P        W  OE    4.15.0-65-generic #74~16.04.1-Ubuntu
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711159] Hardware name: Dell Inc. Inspiron 15 7000 Gaming/065C71, BIOS 1.2.0 10/06/2017
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711161] RIP: 0010:v4l_fill_fmtdesc+0xf4d/0x1260 [videodev]
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711162] RSP: 0018:ffff9c3184157c50 EFLAGS: 00010286
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711162] RAX: 0000000000000000 RBX:ffff9c3184157d68 RCX: 0000000000000006
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711163] RDX: 0000000000000007 RSI:0000000000000082 RDI: ffff8fa63f516490
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711163] RBP: ffff9c3184157c60 R08:0000000000000001 R09: 00000000000004ed
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711164] R10: ffff8fa622f0fb40 R11: 00000000000004ed R12: ffffffffc1b4c860
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711164] R13: ffff8fa622f0fb40 R14: ffff8fa55c512e00 R15: 0000000000000000
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711165] FS:  00007fcae8c2e700(0000) GS:
ffff8fa63f500000(0000) knlGS:0000000000000000
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711165] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711166] CR2: 00007fcae41d9010 CR3: 000000035063e005 CR4: 00000000003606e0
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711166] Call Trace:
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711169]  v4l_enum_fmt+0x7e/0x140 [videodev]
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711171]  __video_do_ioctl+0x2a4/0x320 [videodev]
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711174]  ? video_ioctl2+0x20/0x20 [videodev]
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711176]  video_usercopy+0x285/0x630 [videodev]
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711178]  ? lru_cache_add_active_or_unevictable+0x36/0xb0
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711179]  ? __handle_mm_fault+0xf1e/0x1240
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711182]  video_ioctl2+0x15/0x20 [videodev]
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711184]  v4l2_ioctl+0xe8/0xf0 [videodev]
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711185]  do_vfs_ioctl+0xa4/0x600
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711186]  ? handle_mm_fault+0xcc/0x1f0
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711188]  ? __do_page_fault+0x2ae/0x4d0
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711189]  SyS_ioctl+0x79/0x90
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711190]  do_syscall_64+0x73/0x130
Oct 28 10:12:12 ubuntubot kernel: [ 2226.711191]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2

Oct 28 10:12:12 ubuntubot kernel: [ 2226.711192] RIP: 0033:0x7fcb12e93f47

Oct 28 10:12:12 ubuntubot kernel: [ 2226.711192] RSP: 002b:00007fcae8c2c978 EFLAGS: 00000246 ORIG_RAX: 0000000000000010

Oct 28 10:12:12 ubuntubot kernel: [ 2226.711193] RAX: ffffffffffffffda RBX: 00007fcae4023b20 RCX: 00007fcb12e93f47

Oct 28 10:12:12 ubuntubot kernel: [ 2226.711193] RDX: 00007fcae8c2cb00 RSI: 00000000c0405602 RDI: 0000000000000021

Oct 28 10:12:12 ubuntubot kernel: [ 2226.711194] RBP: 00007fcae8c2cf00 R08: 
00000000000001e0 R09: 0000000000000001

Oct 28 10:12:12 ubuntubot kernel: [ 2226.711194] R10: 00000000000001e0 R11: 0000000000000246 R12: 00007fcae401af10

Oct 28 10:12:12 ubuntubot kernel: [ 2226.711195] R13: 00007fcae8c2cf60 R14: 00007fcae8c2cf00 R15: 00007fcae401ae70

Oct 28 10:12:12 ubuntubot kernel: [ 2226.711195] Code: 30 0f 84 b4 01 00 00 3d 4a 4c 32 30 0f 84 97 01 00 00 3d 4d 33 31 30 0f 84 7a 01 00 00 89 c6 48 c7 c7 b2 10 ac c1 e8 93 45 9e f8 <0f>0b 80 7b 0c 00 0f 85 d6 f3 ff ff 8b 43 2c 48 c7 c1 59 04 ac

``` 

Observing the /var/crash shows me that the crash resulted from the d435 nodelet:
`ProcCmdline: /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/eyal/.ros/log/cdbdc75e-f950-11e9-8f67-7c67a2f01fa4/camera_back-realsense2_camera_manager-7.log
`
What other information can I provide in order to aid in finding a solution? I currently get random system crashes which are a real risk for my robot...

Thanks,
Eyal

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@AutomatoRobotics Could you please specify more about what you ran(such as launch file) when got this issue?

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@AutomatoRobotics  Could you please update? Thanks!

AutomatoRobotics, 2019-12-15:
Hi, sorry for being unresponsive. We're having a series of field test this Wednesday so I'll keep you updated on whether the latest sdk/wrapper updates solved it. If not, I'll send the launch file

AutomatoRobotics, 2019-12-19:
Hello,

Unfortunately it's still problematic, though much less frequent crashes than before.
I noticed that it usually crashes when I kill the node (with ctrl-c). The computer immediately shuts down - no warning at all.

The launch file is attached, all the arguments are set in the beginning as I run it without passing any arguments:
[1.launch.txt](https://github.com/IntelRealSense/realsense-ros/files/3983641/1.launch.txt)

RealSenseCustomerSupport, 2020-02-06:
----------------------------------------------

@AutomatoRobotics Which SDK and ROS wrapper version did you try? Any crash log for killing node operation(Ctrl+C)? Did you get crash when running the node?

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@AutomatoRobotics Any update from your side? Could you please update?

AutomatoRobotics, 2020-02-14:
Sorry for the late reply, since the recent updates we didn't encounter more
crashes.
If the problem resurfaces we'll let you know.

Many thanks!

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@AutomatoRobotics Glad to hear that. Yes, please feel free to let us know if you have any other issues or questions.

closed 2020-02-14 by RealSenseCustomerSupport