realsenseai/realsense-ros #970: Support for D435e

Troski, 2019-10-29:
I am trying to use this ROS driver with a D435e camera from Framos, however is not being detected by the ROS driver. I am however able to get depth and rgb using realsense-viewer. Is GiGe supported by this driver? 

```
marco@marco-Blade:~/catkin_ws/src/realsense-ros/realsense2_camera$ roslaunch realsense2_camera rs_rgbd.launch --screen
... logging to /home/marco/.ros/log/f1189f60-fa01-11e9-9805-00e04c68e7db/roslaunch-marco-Blade-23271.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://marco-Blade:41073/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: camera/odom_in
 * /camera/realsense2_camera/unite_imu_method: none
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [23281]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f1189f60-fa01-11e9-9805-00e04c68e7db
process[rosout-1]: started with pid [23294]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [23311]
process[camera/realsense2_camera-3]: started with pid [23312]
process[camera/color_rectify_color-4]: started with pid [23313]
[ INFO] [1572322186.567799856]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
process[camera/points_xyzrgb_hw_registered-5]: started with pid [23319]
[ INFO] [1572322186.574345842]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1572322186.579886753]: Loading nodelet /camera/color_rectify_color of type image_proc/rectify to manager realsense2_camera_manager with the following remappings:
[ INFO] [1572322186.579940204]: /camera/image_mono -> /camera/color/image_raw
[ INFO] [1572322186.579953426]: /camera/image_rect -> /camera/color/image_rect_color
[ INFO] [1572322186.580256883]: Initializing nodelet with 8 worker threads.
[ INFO] [1572322186.583356727]: Loading nodelet /camera/points_xyzrgb_hw_registered of type depth_image_proc/point_cloud_xyzrgb to manager realsense2_camera_manager with the following remappings:
[ INFO] [1572322186.583405001]: /camera/depth_registered/image_rect -> /camera/aligned_depth_to_color/image_raw
[ INFO] [1572322186.583415686]: /camera/depth_registered/points -> /camera/depth_registered/points
[ INFO] [1572322186.583425139]: /camera/rgb/camera_info -> /camera/color/camera_info
[ INFO] [1572322186.583431659]: /camera/rgb/image_rect_color -> /camera/color/image_rect_color
[ INFO] [1572322186.595945507]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.
[ INFO] [1572322186.661910390]: RealSense ROS v2.2.9
[ INFO] [1572322186.661978710]: Running with LibRealSense v2.29.0
[ WARN] [1572322187.121847432]: No RealSense devices were found!
[ WARN] [1572322193.237008550]: No RealSense devices were found!
[ WARN] [1572322199.353150731]: No RealSense devices were found!
```

MartyG-RealSense, 2019-10-29:
According to a development roadmap that FRAMOS recently shared in a webinar session about D435e, a ROS wrapper for the camera will be provided in firmware version FW 1.3.5 that was due to be released in mid October 2019. 

Page 15 of the D435e user guide document, available from the link below, explains how to perform a firmware update of D435e, on the 'documentation D435e' link.

https://www.framos.com/en/support/downloads/

Because the D435e is a FRAMOS product, technical support should be sought from FRAMOS and not Intel.

Troski, 2019-11-05:
The new driver is now available, and works with the current ROS driver.

closed 2019-11-05 by Troski

antoineTK, 2020-03-09:
Hi, 

I am having the same issue "No RealSense devices were found!" running ros kinetic. Camera is updated to latest 1.5.1 firmware. Anything I can do?

MartyG-RealSense, 2020-03-09:
@antoineTK   If you are using the FRAMOS D435e, hardware tech support for it is the responsibility of FRAMOS as it is not an Intel product.  This is also true for other products that incorporate RealSense technology but are not made by Intel.

FRAMOS tech support can be found here:

https://www.framos.com/en/solutions/technical-support/

maximilianwulf, 2020-07-08:
@antoineTK, ROS wrapper v2.2.9 with FRAMOS realease v1.8.0 works on my machine.

antoineTK, 2020-07-09:
I do not remember which software version I got it to work with, but I do have the FRAMOS working now and it was just a matter of getting the latest release