realsenseai/realsense-ros #971: initial_reset option causes crashes on other driver instances

mikepurvis, 2019-10-29:
As in the title, we get a lot of crashed nodelet managers if we try to launch multiple instances of the driver with `initial_reset:=true`. Errors look like:

```
terminate called after throwing an instance of 'rs2::backend_error'
  what():  map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
terminate called after throwing an instance of 'rs2::backend_error'
  what():  device is no longer connected! Last Error: Success
```

RealSenseCustomerSupport, 2019-11-26:
----------------------------------------------

Hi @mikepurvis,

1. Which camera do you use? How many cameras do you use?
2. What's your launch command?

mikepurvis, 2019-11-26:
4x D435, launch command is irrelevant to this discussion.

Disabling `initial_reset` eliminates this particular crash, and we handled the others by separating our realsense launcher off and repeatedly re-launching it until everything comes up without crashing.

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

@mikepurvis,

Trying to launch 4 cameras with below command on my NUC + Ubuntu 16.04 + Kernel 4.15 has no problem. One powered USB3 hub with 4A supply is used  for three cameras and one USB3 port for another camera.
$ roslaunch realsense2_camera rs_multiple_devices.launch initial_reset:=true serial_no_camera1:=829212072349 serial_no_camera2:=841612071814 serial_no_camera3:=818312071211 serial_no_camera4:=816612061662

I wonder if any power issue since 4 x D435 used? Any trouble if only one or two D435 used?

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

By the way, I use the ros 2.2.9 and librealsense 2.29.0 for the test. All four camera's firmware version is 5.11.15.0.

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

Hi @mikepurvis,

Haven't seen a response on this thread in some time.  Let us know the latest.  If not we can close the thread out.

doronhi, 2019-12-29:
I am aware of an issue regarding using multiple nodes with initial_reset. I think we should keep this one open for a while, hoping I'll get around to investigate some more.

doronhi, 2021-04-14:
It's been a long time. I tested again, with 4 cameras, and could not reproduce.

closed 2021-05-03 by RealSenseSupport