realsenseai/realsense-ros #975: Unable to get color texture for pointcloud2: pointcloud_texture_stream:=RS2_STREAM_COLOR has no effect

iovsiann, 2019-10-31:
When viewing pointcloud2 in rviz the point cloud looks gray, has no color texture from RGB camera. However, when using realsense-viewer with 3D cloud view enabled and RGB selected as texture source - the point cloud looks fine, with color texture applied.

$ roslaunch realsense2_camera rs_camera.launch filters:=colorizer,pointcloud align_depth:=true pointcloud_texture_stream:=RS2_STREAM_COLOR
... logging to /home/user/.ros/log/43854b7e-fb86-11e9-ae86-000c2968ac22/roslaunch-ubuntu-4508.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:44103/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: colorizer,pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 71 lines cut ...]
[ INFO] [1572488970.315658294]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
 30/10 19:29:30,315 WARNING [140079774693120] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1572488970.317394875]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1572488970.320970529]: setupPublishers...
[ INFO] [1572488970.323021752]: Expected frequency for depth = 30.00000
[ INFO] [1572488970.355847209]: Expected frequency for infra1 = 30.00000
[ INFO] [1572488970.381421345]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1572488970.406333614]: Expected frequency for infra2 = 30.00000
[ INFO] [1572488970.433021481]: Expected frequency for color = 30.00000
[ INFO] [1572488970.456496593]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1572488970.484567756]: setupStreams...
[ INFO] [1572488970.486747864]: insert Depth to Stereo Module
[ INFO] [1572488970.486848216]: insert Color to RGB Camera
[ INFO] [1572488970.487029357]: insert Infrared to Stereo Module
[ INFO] [1572488970.487043023]: insert Infrared to Stereo Module
[ INFO] [1572488970.503902861]: SELECTED BASE:Depth, 0
[ INFO] [1572488970.565673878]: RealSense Node Is Up!
 30/10 19:29:30,755 WARNING [140079441815296] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 30/10 19:29:30,963 WARNING [140079670482688] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ERROR] [1572488986.435015992]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
[ERROR] [1572489025.491262531]: An error has occurred during frame callback: Error occured during execution of the processing

====================================

[image]

doronhi, 2019-11-10:
In RViz, in the PointCloud2 options, try changing "Color Transformer" from "Intensity" to "RGB8"

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

Hi @iovsiann,

Is the issue resolved per @doronhi's suggestion? If so, will close this.

closed 2019-12-26 by RealSenseCustomerSupport

bgilreath, 2020-05-23:
> In RViz, in the PointCloud2 options, try changing "Color Transformer" from "Intensity" to "RGB8"

Has no effect. 

Jetson Nano:
Operating System: Ubuntu 18.04.4 LTS
Kernel: Linux 4.9.140-tegra
Architecture: arm64

[ INFO] [1590199616.245470051]: Initializing nodelet with 4 worker threads.
[ INFO] [1590199616.546283534]: RealSense ROS v2.2.13
[ INFO] [1590199616.546376712]: Running with LibRealSense v2.32.1
[ INFO] [1590199616.588678004]:  
[ INFO] [1590199617.047181295]: Device with serial number XXX was found.

[ INFO] [1590199617.047289005]: Device with physical ID 2-1.2-4 was found.
[ INFO] [1590199617.047373121]: Device with name Intel RealSense D435 was found.
[ INFO] [1590199617.048767827]: Device with port number 2-1.2 was found.
[ INFO] [1590199617.048872099]: Resetting device...
[ INFO] [1590199623.164680448]:  
 22/05 19:07:03,357 WARNING [546853343616] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 0, 
 22/05 19:07:03,361 WARNING [546853343616] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 0, 
[ INFO] [1590199623.625840667]: Device with serial number XXX was found.

[ INFO] [1590199623.625971293]: Device with physical ID 2-1.2-4 was found.
[ INFO] [1590199623.626084680]: Device with name Intel RealSense D435 was found.
[ INFO] [1590199623.628561432]: Device with port number 2-1.2 was found.
[ INFO] [1590199623.633198369]: getParameters...
[ INFO] [1590199623.733211319]: setupDevice...
[ INFO] [1590199623.733296216]: JSON file is not provided
[ INFO] [1590199623.733347831]: ROS Node Namespace: camera
[ INFO] [1590199623.733546115]: Device Name: Intel RealSense D435
[ INFO] [1590199623.733777420]: Device Serial No: XXX
[ INFO] [1590199623.733827577]: Device physical port: 2-1.2-4
[ INFO] [1590199623.734028830]: Device FW version: 05.12.03.00
[ INFO] [1590199623.734193832]: Device Product ID: 0x0B07
[ INFO] [1590199623.734337532]: Enable PointCloud: On
[ INFO] [1590199623.734410085]: Align Depth: Off
[ INFO] [1590199623.734474356]: Sync Mode: On
[ INFO] [1590199623.734610764]: Device Sensors: 
[ INFO] [1590199623.734729985]: Stereo Module was found.
[ INFO] [1590199623.734826496]: RGB Camera was found.
[ INFO] [1590199623.734965144]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1590199623.735135303]: Add Filter: decimation
[ INFO] [1590199623.736527769]: Add Filter: colorizer
[ INFO] [1590199623.738470972]: Add Filter: pointcloud
[ INFO] [1590199623.739847189]: num_filters: 3
[ INFO] [1590199623.739929898]: Setting Dynamic reconfig parameters.
[ INFO] [1590199629.155015965]: Done Setting Dynamic reconfig parameters.
[ INFO] [1590199629.204120163]: depth stream is enabled - width: 424, height: 240, fps: 15, Format: Z16
[ INFO] [1590199629.204464231]: infra1 stream is enabled - width: 424, height: 240, fps: 15, Format: Y8
[ INFO] [1590199629.204778766]: infra2 stream is enabled - width: 424, height: 240, fps: 15, Format: Y8
[ INFO] [1590199629.239839339]: color stream is enabled - width: 424, height: 240, fps: 15, Format: RGB8
[ INFO] [1590199629.239963195]: setupPublishers...
[ INFO] [1590199629.246431250]: Expected frequency for depth = 15.00000
[ INFO] [1590199629.321799707]: Expected frequency for infra1 = 15.00000
[ INFO] [1590199629.365351592]: Expected frequency for infra2 = 15.00000
[ INFO] [1590199629.409614684]: Expected frequency for color = 15.00000
[ INFO] [1590199629.456985578]: setupStreams...
[ INFO] [1590199629.561984171]: insert Depth to Stereo Module
[ INFO] [1590199629.562146829]: insert Color to RGB Camera
[ INFO] [1590199629.562232299]: insert Infrared to Stereo Module
[ INFO] [1590199629.562309852]: insert Infrared to Stereo Module
[ INFO] [1590199629.837427283]: SELECTED BASE:Depth, 0
[ INFO] [1590199629.911971354]: RealSense Node Is Up!