realsenseai/realsense-ros #976: Error while loading image - libusb_bulk_transfer failed. LIBUSB_ERROR_CODE: -1 LIBUSB_ERROR_IO

xiongdahua555, 2019-10-31:
Required Info |  
-- | --
Camera Model | { T265 }
Firmware Version | 0.0.18.6100
Operating System & Version | ubntu18.04
Kernel Version (Linux Only) | 4.4.167
Platform | RK3399pro
SDK Version | 2.25.0
Language | C++
Segment | Robot

Issue Description
using ros package which you provide，it work well when i run the launch firstly.But when i terminate package using ctrl+c and rerun the package，it print error：Error while loading image - libusb_bulk_transfer failed. LIBUSB_ERROR_CODE: -1 LIBUSB_ERROR_IO

doronhi, 2019-11-10:
By default librealsense2 uses video4linux for connection with the device. If you specify "FORCE_LIBUVC" while building, librealsense2 will use libuvc.
libuvc does not take care of the camera in case of exception in the user space. It does not clean the memory, close the device etc. Therefor, after ctrl-c is pressed, the device is in unstable status.
video4linux does take care of releasing the device in such cases.
If it is possible for you to use video4linux with your platform and OS, it may solve the issue.

RealSenseCustomerSupport, 2020-01-10:
----------------------------------------------

Will you be needing further assistance with this?

closed 2020-01-20 by RealSenseCustomerSupport