realsenseai/realsense-ros #977: How to convert _d435.urdf.xacro  into urdf

gachiemchiep, 2019-10-31:
Hello guys.

I know this is a noob question. But how can i convert the _d435.urdf.xacro  into urdf ?
I try the following command

```bash
rosrun xacro xacro --inorder src/realsense-ros/realsense2_description/urdf/_d435.urdf.xacro 
```
The output is as follow

```bash
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from src/realsense-ros/realsense2_description/urdf/_d435.urdf.xacro | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<!--
License: Apache 2.0. See LICENSE file in root directory.
Copyright(c) 2017 Intel Corporation. All Rights Reserved

This is the URDF model for the Intel RealSense 430 camera, in it's
aluminum peripherial evaluation case.
-->
<robot name="sensor_d435" xmlns:xacro="http://ros.org/wiki/xacro">
</robot>

```

I can't include into my urdf. 
So has anyone met the same problem ? And how can i solve it.

closed 2019-11-02 by gachiemchiep

q8wwe, 2020-03-04:
How did you solve the problem

gachiemchiep, 2020-03-04:
@q8wwe  
we didn't solve the problem so we stop using the urdf format and convert everything into xacro format.

q8wwe, 2020-03-04:
Did you ever simulate realsense cam with gazebo ?

gachiemchiep, 2020-03-04:
@q8wwe 
No we use rviz for visualizing. 
Sorry, we couldn't share the file

q8wwe, 2020-03-04:
So you aren't using gazebo just rviz.which mean you ate working on real robot not simulator

gachiemchiep, 2020-03-04:
@q8wwe 
That's true. we're working on the UR5