realsenseai/realsense-ros #979: after run realsense-viewer ,show this question

Grandzxw, 2019-11-01:
it happens:
01/11 19:56:12,567 WARNING [139852830111488] (types.cpp:48) hwmon command 0x4f failed. Error type: No data to return (-21).
01/11 19:56:12,735 WARNING [139852830111488] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
01/11 19:56:17,799 WARNING [139852830111488] (types.cpp:48) set_pu(id=3) failed! Last Error: Input/output error
01/11 19:56:17,799 WARNING [139852830111488] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
01/11 19:56:22,838 WARNING [139852830111488] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
01/11 19:56:22,839 WARNING [139852830111488] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
01/11 19:56:52,860 WARNING [139852695717632] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
01/11 19:57:07,098 WARNING [139852673361664] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.

doronhi, 2019-11-10:
Hello,
Sorry but it is not clear to me what are trying to run and is not working. realsense-viewer?
What operation system? what versions for librealsense2 and realsense2_camera?
What is the command line you run? How did you install the app?
Sorry for the questions, it is important to make sure I understand the issue.

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

Hi @Grandzxw,

Any update?

closed 2019-12-26 by RealSenseCustomerSupport