realsenseai/realsense-ros #980: ddynamic_reconfigure package issue!

netgittemp, 2019-11-05:
Hi,
I am following the install instruction.
However, when I compile it, I have the following error;
```
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "ddynamic_reconfigure" with any of the following names:

    ddynamic_reconfigureConfig.cmake
    ddynamic_reconfigure-config.cmake

  Add the installation prefix of "ddynamic_reconfigure" to CMAKE_PREFIX_PATH
  or set "ddynamic_reconfigure_DIR" to a directory containing one of the
  above files.  If "ddynamic_reconfigure" provides a separate development
  package or SDK, be sure it has been installed.
```

So, I tried to install ddynamic_recofigure using
`$ sudo apt install ros-kinetic-ddynamic-reconfigure`
However, I got the following error as well.
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-ddynamic-reconfigure
```

Please help me to install this package.....

JadTawil-theonly, 2019-11-06:
i think you have to clone this into your catkin workspace:
https://github.com/pal-robotics/ddynamic_reconfigure

This is similar to dynamic reconfigure, but allows you to modify parameters without a cfg file.

G-mel, 2019-11-06:
make sure you have everything [here](https://github.com/IntelRealSense/realsense-ros/blob/development/.travis.yml) installed. For me, it was a matter of running

```
sudo apt install ros-<distro>-ddynamic-reconfigure
```

where my distro is `melodic`

doronhi, 2019-11-10:
You should only download the code and compile ddynamic-reconfigure by yourself if you failed to install the debian package, as you did, but I would first try to figure out why that installation failed.
The "ddynamic-reconfigure" is a ROS distributed debian package so, assuming you run Ubuntu 16 and ROS Kinetic, I would start at the beginning:
Did you follow the ROS installation procedure [here](http://wiki.ros.org/kinetic/Installation/Ubuntu) - adding "packages.ros.org" to the sources.list?
Is `sudo apt-get update` working ok?

netgittemp, 2019-11-11:
I do not what was the problem..
but now `$ sudo apt install ros-kinetic-ddynamic-reconfigure` is working.
What I did is only following (https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages)

@doronhi I installed ROS correctly. Absolutely, I added "packages.ros.org" to the sources.list correctly and `sudo apt-get update` is working correctly.

Thank you anyway.

P.S. I will reinstall ubuntu and try from the first step.
I will let you know if I find any problem.

closed 2019-11-12 by netgittemp

YoshuaNava, 2020-01-29:
@netgittemp I'm sorry for re-opening this issue, but my question is closely related to what you did to solve this issue.

How were you able to solve the problem with the Debian package? The API of the ddynamic_reconfigure official Debian [[ROS package index](https://index.ros.org/p/ddynamic_reconfigure/#melodic)], something that you can see by looking at the version (0.2.0)

The version of ddynamic_reconfigure that realsense-ros requires is 0.2.2. This one is not available as a Debian, but only source here: https://github.com/pal-robotics/ddynamic_reconfigure

How were you able to solve the problem by installing the Debian? You must have a different ROS mirror configured or built the package from source. If I install that Debian I get build errors due to the API expected by realsense-ros being completely different.

chfritz, 2020-06-01:
For anyone coming to this issue when trying to build on ROS noetic (ubuntu 20.04) like me: I got the same error as the OP and noticed that the ddynamic-reconfigure package was not available in the apt repo of noetic (yet?). But cloning the source package into my catkin workspace, as [suggested above](https://github.com/IntelRealSense/realsense-ros/issues/980#issuecomment-550338910), worked.