realsenseai/realsense-ros #981: Setting Disparity Shift

JadTawil-theonly, 2019-11-06:
Hello, 

I would like to configure the disparity shift (value from 0 -> 255), in order to control the minZ value of the depth sensing.

How can i do this via the ros wrapper? 

Would changing this require a recalibration?

doronhi, 2019-11-10:
"Advanced Controls" are currently not supported directly by the ros wrapper.
You can, however, modify them on start up by setting a json file.
To create a configuration json file for the first time use realsense-viewer and the "export" preset button:
[image]
To start realsense2_camera with the preset file use the "json_file_path" parameter:
`roslaunch realsense2_camera rs_camera.launch json_file_path:=/home/doronhi/catkin_ws/conf.json`

To control the disparity shift modify value for the field "param-disparityshift" in the json file.

Changing this does not require re-calibration.

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@JadTawil-theonly Did you try the suggestion from doronhi? Any update? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@JadTawil-theonly Could you please update? Thanks!

JadTawil-theonly, 2019-12-13:
Hey!
Yes is all good

closed 2019-12-13 by JadTawil-theonly