realsenseai/realsense-ros #982: Error while run rs_camera (mismatch error)

ARLVOL, 2019-11-07:
Hi,

I just install the latest sdk (2.30) and now I can't run nothing on ros because the realsense-ros version is in 2.29 version. How I can solve this?

`Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: API version mismatch: librealsense.so was compiled with API version 2.30.0 but the application was compiled with 2.29.0! Make sure correct version of the library is installed (make install)
`

Tnx
Aviad

doronhi, 2019-11-10:
librealsense 2.30 is compatible with librealsense 2.29.
Assuming you are using realsense2_camera's latest version (v2.2.9), you just need to recompile it with librealsense2 2.30.

ARLVOL, 2019-11-10:
Hi,
I installed the latest sdk (2.30) and then I clone the latest realsense-ros (v2.2.9) to /src folder. 
I tried many combination cut I always get this error. 
I am doing something wrong? 
I did these steps many time and I don't know why this time it doesn't work.
I am trying now again and I will update.

Thanks

ARLVOL, 2019-11-10:
Hi,

I don't know what I did but now it is working.
I did again what I did before.

Thanks anyway :)

closed 2019-12-01 by RealSenseCustomerSupport