realsenseai/realsense-ros #983: T265 imu and gyro rates are opposite of the values listed in the datasheet

stevendaniluk, 2019-11-09:
In table 3.3 on page 10 of the [T265 datasheet](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/IntelRealSenseTrackingT265Datasheet.pdf) the gyro sample rate is listed as 62.5 Hz, while the accelerometer sample rate is listed as 200 Hz.

[image]

However, in the [T265 launch file](https://github.com/IntelRealSense/realsense-ros/blob/2.2.9/realsense2_camera/launch/rs_t265.launch#L22) (taken from release 2.2.9) the accelerometer update rate is defined as 62 Hz, while the gyro update rate is listed as 200 Hz.

Can someone please provide some insight into which one is correct?

If you try to reverse these rates in the launch file (set the accelerometers to 200 Hz and the gyro to 62 Hz) you will get the following warnings:
```
[ WARN] [1573262579.793574359]: No mathcing profile found for gyro with fps=62
[ WARN] [1573262579.793636393]: profiles found for gyro:
[ WARN] [1573262579.793659110]: fps: 200. format: MOTION_XYZ32F
[ WARN] [1573262579.793701541]: No mathcing profile found for accel with fps=200
[ WARN] [1573262579.793706988]: profiles found for accel:
[ WARN] [1573262579.793734800]: fps: 62. format: MOTION_XYZ32F
```

RealSenseCustomerSupport, 2019-11-27:
----------------------------------------------

Hi Steven,

The datasheet you're referencing below appears to be an outdated version. You'll find the latest here (with the correct values.)

https://dev.intelrealsense.com/docs/tracking-camera-t265-datasheet

Thanks

stevendaniluk, 2019-12-02:
Thank you. I initially had some difficulty finding the datasheet via Intel's website and had resorted to Google, which apparently led me to an old version. Glad this was fixed.

closed 2019-12-02 by stevendaniluk