realsenseai/realsense-ros #986: error: [Errno 11] Resource temporarily unavailable

SidKuruba, 2019-11-12:
sikm@dhcp-10-19-65-1[11/12 17:05]:catkin_ws$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/sikm/.ros/log/902bac8e-0540-11ea-bbdf-b06ebfcd6686/roslaunch-dhcp-10-19-65-1-8308.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 471, in load_config_default
    config.assign_machines()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 222, in assign_machines
    if [m for m in machine_unify_dict.values() if not is_machine_local(m)]:
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/core.py", line 94, in is_machine_local
    machine_ips = [host[4][0] for host in socket.getaddrinfo(machine.address, 0, 0, 0, socket.SOL_TCP) if isinstance(host[4][0], str)]
error: [Errno 11] Resource temporarily unavailable

SidKuruba, 2019-11-12:
I have the realsense camera d3451 connected and have followed the steps at 
https://github.com/IntelRealSense/realsense-ros/blob/development/README.md

But ran into the above issue when I ran the command - "roslaunch realsense2_camera rs_camera.launch"

Any inputs would be helpful.

This is on Ubuntu 16.04 LTS

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

Hi @SidKuruba,

It looks the error happens in the very beginning about starting roslaunch server.

doronhi, 2019-12-05:
I found [this reference](https://www.bountysource.com/issues/42538310-shell-and-or-mavros-node-dies-with-serial-connection)
Maybe the ROS port is already taken?
In any case, I think this is not a realsense2_camera issue.

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@SidKuruba,

Please clarify if your setup has some trouble there as @doronhi mentioned.

closed 2019-12-25 by RealSenseCustomerSupport