realsenseai/realsense-ros #987: Standard settings for D435i to get synchronized IMU and color image stream?

levenberg, 2019-11-12:
I am trying to use D435i to run the state-of-the-art visual-inertial odometry system. However, there is always something wrong to obtain good data input. One big issue is the synchronization of IMU and color image stream.  After setting `enable_sync=true`, we can get an imu topic: `\color\imu`. However, the rate of this imu topic is so confusing. 

if we set `unite_imu_method=copy`, the imu rate is constant, can be 263 hz, 450hz, 463hz or 650 hz.
if we set `unite_imu_method=linear_interpolation`, the imu rate is always changing. 

Based on these two settings, I cannot see that the IMU is synchronized to the color image stream, because it is always changing. And there seems no way to compare the timestamp of each topic to compute the lag because the timestamp seems to be added by my local computer, not by the realsense itself.

So is there any standard setting for D435i for using the IMU to get high-quality synchronized data to the color image steam. Can you give us a detailed tutorial on how to use it?

doronhi, 2019-11-13:
Just a clarification: 'enable_sync' causes the image streams to be synchronized but has no effect on the imu. The united imu topic is created by setting the `unite_imu_method`.

That said, there is an explanation on the expected frame rates of the imu in #729.
The Timestamps of both the images and imu are set based on the time stamp given on the device itself. They are not set by the time of arrival to the host computer. Therefor it should be possible to use them for synchronization of IMU and images. However, there is an known issue of different and small latency for each stream that is not fully taken into account yet and should be in upcoming versions.

levenberg, 2019-11-13:
@doronhi Thank you for your clarification. Because I always get the warnings of 
`HID timestamp not found, switching to Host timestamps.` and `Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)`. But I can get a normal imu stream and color image stream, so I guess it is set as the host timestamps. 

Then for the synchronization, if both of the imu and color image stream are uniformed sampled based on the device itself, there should be like a constant time lag between the closed two topics. However, when I use [kalibr](https://github.com/ethz-asl/kalibr) to calibrate the extrinsics between the two sensors by recording the IMU topic (263hz) and color image topic (30hz),   the output is like `the two sensors are not synchronized`.

Grandzxw, 2019-11-15:
> @doronhi Thank you for your clarification. Because I always get the warnings of
> `HID timestamp not found, switching to Host timestamps.` and `Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)`. But I can get a normal imu stream and color image stream, so I guess it is set as the host timestamps.
> 
> Then for the synchronization, if both of the imu and color image stream are uniformed sampled based on the device itself, there should be like a constant time lag between the closed two topics. However, when I use [kalibr](https://github.com/ethz-asl/kalibr) to calibrate the extrinsics between the two sensors by recording the IMU topic (263hz) and color image topic (30hz), the output is like `the two sensors are not synchronized`. The Warning is here:
> `[ INFO] [1573666068.717761557]: RealSense Node Is Up! 13/11 12:27:48,749 WARNING [140220741707520] (sensor.cpp:1154) HID timestamp not found, switching to Host timestamps. [ WARN] [1573666068.749476653]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME) 13/11 12:27:48,828 WARNING [140220733314816] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details. 13/11 12:27:48,837 WARNING [140220716529408] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.`
> 
> When I tested the IMU topic timestamps, my code is here:
> void imu_callback(const sensor_msgs::ImuConstPtr &imu_msg)
> {
> if (imu_msg->header.stamp.toSec() <= last_imu_t)
> {
> ROS_WARN("imu message in disorder!");
> return;
> }
> last_imu_t = imu_msg->header.stamp.toSec();
> }
> It always gives me the warning of "imu message in disorder!", which means the current imu timestamp is equal or even before the last one. It is so weird.

I have the same question :" imu  message in disorder". have you solve it?

levenberg, 2019-11-15:
Hi @Grandzxw , I have not solved it yet. And I found the timestamps of subsequent IMU topics sometimes is not only equal to the `last_imu_t`, but smaller than the former timestamp, it is so weird. Then the color camera seems to be a rolling shuttle camera, I am trying to use the infrared camera ( global shutter)instead. But for D435i, I tried to use `realsense_viewer` to turn off the IR projector. However, it shows: 
```
Invalid Value in rs2_set_option(options:0x7f1ebc010d40, option:Visual Preset, value:6):
apply_preset(...) failed! Given device doesn't support Remove IR Pattern Preset (pid=0xb3a).
```

@doronhi Could you please tell me how to turn off the IR projector for D435i?

doronhi, 2019-11-17:
You are mixing issues. It would be better, I think, to open a new issue for each topic.
As for the disorder of the timestamps, Thank you for pointing this out to me. I was aware of an issue with the T265 but I'll have to look into the problem in the D435i device.

As for the IR projector, you should turn the option "emitter_enabled" off. 
The visual preset "remove IR pattern" is a filter for using on D415 and the like for removing the dots while receiving synthetic RGB images out of IR images.

Grandzxw, 2019-11-18:
> You are mixing issues. It would be better, I think, to open a new issue for each topic.
> As for the disorder of the timestamps, Thank you for pointing this out to me. I was aware of an issue with the [...]
 **and I find the  timestamps of subsequent IMU topics sometimes is   equal to the last_imu_t sometime ,and I read it in PC ,so I want to ask  whether is it related to computer     @@@@hardware.**

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

Has this PR been tried for this issue:  https://github.com/IntelRealSense/realsense-ros/pull/1010

levenberg, 2019-12-14:
Currently, I am using librealsense version: 2.17.1 and realsense ros version 2.1.3. And this old version provides good timestamps. So I have not tried the latest update.

RealSenseCustomerSupport, 2019-12-16:
----------------------------------------------

Maybe we should close this thread out until you're able to test out the latest release and provide some update.

Let me know if you're inline with this direction.

Thanks

levenberg, 2019-12-18:
Thank you for your update. I have updated to the latest version `RealSense ROS v2.2.11` and ` LibRealSense v2.31.0`. However, the IR projector turn off command does not work any more.
```
<rosparam>
  /camera/stereo_module/emitter_enabled: false
</rosparam>
```
Duplicate of #817

DekkerQ, 2020-03-27:
> Thank you for your update. I have updated to the latest version `RealSense ROS v2.2.11` and ` LibRealSense v2.31.0`. However, the IR projector turn off command does not work any more.
> 
> ```
> <rosparam>
>   /camera/stereo_module/emitter_enabled: false
> </rosparam>
> ```
> 
> Duplicate of #817 .
> 
> I use `rosrun rqt_reconfigure rqt_reconfigure` to check the value of emitter_enabled now. It has three options `0, 1,2`. So the setting:
> 
> ```
> <rosparam>
>   /camera/stereo_module/emitter_enabled: 0
> </rosparam>
> ```
> 
> works. And I also checked the timestamp, it fine now for running VINS-Mono use rgb camera or infrared camera with IMU.

Hi  @levenberg 

We faced the same issue with the unsynchronized RGB and IMU streams for our viSLAM project and ended up switching to the IR for now. I was wondering if you managed to find a solution for obtaining synchronized IMU and RGB data or that you sticked with the IR as well. 

Thanks

levenberg, 2020-03-31:
Hi @DekkerQ 
If you are using the latest version of  librealsense and realsense-ros, you can directly run viSLAM on the (rolling shutter) RGB camera, but if you want to use one camera of the stereo module (global shutter), IR need to be turned off for feature extraction. If you want to keep the IR on, one way is double the imager frequency and on-off the IR projector, so both good depth and pure image can be obtained. One another option is using the old version: librealsense 2.17.1 and realsense ros version 2.1.3. They are tested quite well.

RealSenseCustomerSupport, 2020-06-03:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-06-12 by RealSenseCustomerSupport

Joeyyuenjoyslife, 2020-08-11:
> Thank you for your update. I have updated to the latest version `RealSense ROS v2.2.11` and ` LibRealSense v2.31.0`. However, the IR projector turn off command does not work any more.
>
> ``` [...]

Hi,@levenberg.  Thank you for your sharing.  has there any  chance that you still remmember the version of Realsense Firmware used in yout test?  I have tried 'LibRealSense V2.17.1' with 'RealSense ROS v2.1.3' , and also 'LibRealSense V2.31.0' with   'RealSense ROS v2.2.11'.  Both failed while running VINS-MONO with warning of “imu msg misorder &&  throw img”。 and I have tried both RGB and IR(projector off) images and didin't work well. 
Thanks

levenberg, 2020-08-11:
Hi @Joeyyuenjoyslife,  
The setting of librealsense 2.17.1 and realsense-ros version 2.1.3 works for me. Maybe you need to check the IMU settings in rs_camera.launch. By the way, you can firstly test VINS-Mono on my collected data using realsense ([Drive](https://drive.google.com/file/d/1Ffke-wFpRifmw7nvcwQCreFm1S7fI3XR/view?usp=sharing)). It uses the left IR imager and IMU.

Joeyyuenjoyslife, 2020-08-12:
> Hi @Joeyyuenjoyslife,
> The setting of librealsense 2.17.1 and realsense-ros version 2.1.3 works for me. Maybe you need to check the IMU settings in rs_camera.launch. By the way, you can firstly test [...]

Hi,@levenberg 
Thanks alot for your help. I have tried your corridors.bag  in Vins-mono and it works great(with my own device's calibration file). but when I use my own d435i, the imu msg in disorder keeps coming up . So below is my rs_camera.launch , is there anything I should modified? if all the above seems normal, I believe the only problem maybe the device it self or the device firmware versions we used are different.

There is something also quite strange,  I downloaded the realsense-ros 2.1.3 from the official repo but when I launch rs_camera.launch , the log shows the version of the realsense-ros is 2.1.2 . Have you experienced the same situation?  or it just dosen't matter?

```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="true"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra1_width"        default="640"/>
  <arg name="infra1_height"       default="480"/>
  <arg name="enable_infra1"       default="true"/>

  <arg name="infra2_width"        default="640"/>
  <arg name="infra2_height"       default="480"/>
  <arg name="enable_infra2"       default="true"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra1_fps"          default="30"/>
  <arg name="infra2_fps"          default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_imu"          default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"           default="true"/>
  <arg name="align_depth"           default="true"/>

  <arg name="filters"               default=""/>
  <arg name="clip_distance"         default="-2"/>
  <arg name="linear_accel_cov"      default="0.01"/>
  <arg name="initial_reset"         default="false"/>
  <arg name="unite_imu_method"      default="linear_interpolation"/>
 
  
  <group ns="$(arg camera)">
   <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra1_width"             value="$(arg infra1_width)"/>
      <arg name="infra1_height"            value="$(arg infra1_height)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>

      <arg name="infra2_width"             value="$(arg infra2_width)"/>
      <arg name="infra2_height"            value="$(arg infra2_height)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra1_fps"               value="$(arg infra1_fps)"/>
      <arg name="infra2_fps"               value="$(arg infra2_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_imu"               value="$(arg enable_imu)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
    </include>
  </group>
</launch>
```

levenberg, 2020-08-12:
@Joeyyuenjoyslife Firstly, my setting of IMU is:
```
<arg name="gyro_fps"            default="200"/>
<arg name="accel_fps"           default="63"/>
<arg name="unite_imu_method"          default="copy"/>
```
But I do not think changing these parameters can correct the disorder issue. I guess you need to redownload the ` realsense-ros 2.1.3`, check the version carefully. Maybe there is an alternative download to make sure the log shows the version of 2.1.3. The version of librealsense  should be matched with the right version of realsense-ros.

ishanivyas, 2020-08-23:
@levenberg  and @Joeyyuenjoyslife 
My goal is to get a calibrated and synchronized IMU & color camera, I do not need any IR (camera or emitter) and do not need the stereo stream produced by those sensors.  I installed realsense-ros v2.2.15 ([1]) with the required version of librealsense (v2.36.0).

Is the ROS launch config in [2] sufficient to publish synchronized IMU and color topics?  I noticed that the new version of realsense-ros splits the "enable_imu" arg into two: "enable_gyro" and "enable_accel", so I have set both to "true" and removed "enable_imu".

[1] https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.15
[2]  https://github.com/IntelRealSense/realsense-ros/issues/987#issuecomment-672507674

Joeyyuenjoyslife, 2020-08-24:
> @Joeyyuenjoyslife Firstly, my setting of IMU is:
>
> ```
> <arg name="gyro_fps"            default="200"/>
> <arg name="accel_fps"           default="63"/>
> <arg name="unite_imu_method" [...]

Hi, @levenberg , I finally made it worked with librealsense 2.31.0 and realsense-ros 2.2.11. But in this test , I updated my realsense firmware to 5.12.01.  thanks a lot .

Joeyyuenjoyslife, 2020-08-24:
> @levenberg and @Joeyyuenjoyslife
> My goal is to get a calibrated and synchronized IMU & color camera, I do not need any IR (camera or emitter) and do not need the stereo stream produced by those sensors. I installed realsense-ros v2.2.15 ([1]) with the required version of librealsense (v2.36.0).
> 
> Is the ROS launch config in [2] sufficient to publish synchronized IMU and color topics? I noticed that the new version of realsense-ros splits the "enable_imu" arg into two: "enable_gyro" and "enable_accel", so I have set both to "true" and removed "enable_imu".
> 
> [1] https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.15
> [2] [#987 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/987#issuecomment-672507674)
> 
> Thanks in advance for your help.

Hi, @ishanivyas . I have not test librealsense2.36.0 and realsense-ros 2.2.15. But I believe it will worked.  If you are using ubuntu 16.04, The other option is that you can use librealsense 2.31.0 and realsense-ros 2.2.11 , and flash your realsense's firmware to version 5.12.01. I have tested it and it works well. if you are using Ubuntu 18.04 , then just make sure your realsense's firmware can satisfied the requirement of the librealsense and realsense-ros.

ishanivyas, 2020-08-25:
Hi, @Joeyyuenjoyslife  Thank you for your quick response. I have double checked the realsense firmware and satifiying other requirements by following this[1]. When I run the command on terminal roslaunch realsense2_camera rs_camera.launch after modifying the rs_camera.launch. I am not able to see camera image or depth image on rviz. Do you have encounter this issue?

 1] https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.15

Thanks in advance for your help.