realsenseai/realsense-ros #988: D435 Depth to Color Extrinsics

rlustig-gpfw, 2019-11-12:
I'd like to get the depth-to-color extrinsics for the D435 camera. Using ROS, I subscribe to `/camera/extrinsics/depth_to_color` to get the extrinsics. However, the returned extrinsics do not make sense:
```
└─▪ rostopic echo /camera/extrinsics/depth_to_color
header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "depth_to_color_extrinsics"
rotation: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
translation: [0.0, 0.0, 0.0]
---
```
The RGB camera and depth sensors are obviously not co-located so I expect to see a translation and slight rotation offset. The `/tf_static` topic also gives the same values.

Am I misunderstanding something here? Using ROS, how can I obtain the depth-to-color extrinsics?

doronhi, 2019-11-13:
You understand correctly and the topic should give you correct values:
```
└─▪ rostopic echo /camera/extrinsics/depth_to_color 
header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "depth_to_color_extrinsics"
rotation: [0.9999784231185913, -0.005351431202143431, -0.003812395967543125, 0.005344766657799482, 0.9999841451644897, -0.001756273559294641, 0.003821734106168151, 0.0017358593177050352, 0.9999911785125732]
translation: [0.014877982437610626, -1.66590980370529e-05, 0.0005684288335032761]
---
```
So the problem seems to be elsewhere. What other information can you supply? What OS do you use? Camera? firmware? Do you see any warnings or errors from the realsense2_camera node?

rlustig-gpfw, 2019-11-13:
Looks like it was a parameter configuration issue. If the `align_depth` parameter is set to `True`, then the depth-to-color extrinsics will always produce the identity extrinsics (you can see this in the code too).

Now, if I set the `align_depth` parameter to `False`, I get expected values:
```
└─▪ rostopic echo /camera/extrinsics/depth_to_color
header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "depth_to_color_extrinsics"
rotation: [0.9999064803123474, 0.012444802559912205, -0.0056675695814192295, -0.012437984347343445, 0.9999218583106995, 0.0012365563306957483, 0.005682515446096659, -0.0011659475276246667, 0.9999831914901733]
translation: [0.014900846406817436, 0.00024103511532302946, 0.00026761708431877196]
---
```

Thanks for the help!

closed 2019-11-13 by rlustig-gpfw