realsenseai/realsense-ros #989: API mismatch problem

OnesoulKang, 2019-11-13:
Hi I'm using Ubuntu 18.04 and ROS melodic

I updated librealsense to v 2.30.0 and then I removed older version realsense-ros and insatlled v 2.2.9.

And I followed the instruction from https://github.com/IntelRealSense/realsense-ros.

After that, I tried to `roslaunch realsense2_camera rs_camera.launch`
But I got this error messages

```
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [24798]
ROS_MASTER_URI=http://192.168.0.22:11311/

setting /run_id to 39d9c65c-05dd-11ea-8f88-9c5c8e76f206
process[rosout-1]: started with pid [24809]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [24816]
process[camera/realsense2_camera-3]: started with pid [24817]
[ INFO] [1573625831.077817514]: Initializing nodelet with 4 worker threads.
[ERROR] [1573625831.172347432]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: API version mismatch: librealsense.so was compiled with API version 2.30.0 but the application was compiled with 2.20.0! Make sure correct version of the library is installed (make install)
[ERROR] [1573625831.172394402]: The error before refreshing the cache was: API version mismatch: librealsense.so was compiled with API version 2.30.0 but the application was compiled with 2.20.0! Make sure correct version of the library is installed (make install)
[FATAL] [1573625831.172623921]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-3] process has died [pid 24817, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/onesoulkang/.ros/log/39d9c65c-05dd-11ea-8f88-9c5c8e76f206/camera-realsense2_camera-3.log].
log file: /home/onesoulkang/.ros/log/39d9c65c-05dd-11ea-8f88-9c5c8e76f206/camera-realsense2_camera-3*.log
```

'realsense-viewer' is working properly.

I tried remove and reinstall several times, but not fixed....

Please let me know what I have to do.

doronhi, 2019-11-13:
It seems that you still have librealsense v2.20.0 somewhere on your system and realsense2_camera was linked with it while building.

OnesoulKang, 2019-11-13:
How can I remove that librealsense v2.20.0?

I already tried this https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

FrederikHaa, 2019-11-21:
Have you rebuild the ROS package? I usually get that error when i update librealsense and don't rebuild the ROS package.

Try to run

`catkin clean realsense2_camera`

and then

`catkin build`

RealSenseCustomerSupport, 2020-01-15:
----------------------------------------------

@OnesoulKang Is the issue resolved? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@OnesoulKang Could you please update?If we don’t hear from you in 7 days, this issue will be closed. Thanks!

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@OnesoulKang Sorry we didn't hear from you for weeks. Will close the ticket at this point. Please feel free to create another ticket if you still have questions or issues. Thanks!

closed 2020-02-17 by RealSenseCustomerSupport