realsenseai/realsense-ros #990: ROS apt package gets out of sync with librealsense2-dev

ihadzic, 2019-11-14:
I have code with build dependency on `librealsense2-dev`, which includes `rs.h`. header file from that library. On the same machine I also have (and use) `ros-kinetic-realsense2-camera` ROS package. Both packages installed using apt. It turns out that the ROS package distributes `rs.h` file that gets placed under /opt/ros/kinetic/include/librealsense2 which is also distributed by librealsese2-dev that places the same file under `/usr/include/librealsense2/rs.h`. In my build environment the ROS path comes first, so the file that actually gets included is the one under /opt/ros. That was not the problem until recently because the two files were the same file, but `librealsense2-dev` 2.30.0 was recently released and the two files diverged. So after updating my system, I ended up with the situation that my application gets tagged that it's using 2.25.1 API, but it's actually build using 2.30.0 library.  The application refuses to run because of this mismatch. I can't take ROS path out of the include-search path because the application I am building is also a ROS node, so it needs everything that catkin sets up for it (which includes all include directories provided by ROS). 

The hack I have to do is actually remove `/opt/ros/include/librealsense2` directory and then it builds fine, picking up include files from `/usr/include/librealsense2`. I don't think that ROS apt package should be distributing headers (nor .so files, which it also does) from foreign libraries, bug instead it should specify the appropriate build and run-time dependencies.

doronhi, 2019-11-17:
Right now there is a bit of a mixup - I'll try to it clear up:
In order to add realsense2_camera to ros-distro (so that `apt-get install ros-kinetic-realsense2-camera` will work), librealsense2 must be available too. Right now it is available on Intel's servers only, hence the registration phase in the librealsense installation manual [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages).
So librealsense version 2.25.1 was added to ros-distro, alongside realsense2_camera version v2.2.6.
This debian has some problems and is not yet maintained. It is not mentioned in the installation guide of realsense2_camera and although it seems to work, there are some known issues with it.
I hope to get around to it soon and have a debian package fully maintained and updated. 
Until then you should remove this installation and install realsense2_camera according to the installation instructions [here](https://github.com/IntelRealSense/realsense-ros)

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

Hi @ihadzic

Were you able to follow the instructions provided by doronhi?  Are you seeing any further issues on this thread?

ihadzic, 2019-12-29:
> Hi @ihadzic
> 
> Were you able to follow the instructions provided by doronhi? Are you seeing any further issues on this thread?

Yes, we are now building the ROS node from the source and installing the librealsense2-dev Debian. There are no more conflicts in .h files.

closed 2019-12-30 by RealSenseCustomerSupport