realsenseai/realsense-ros #991: realsense node crash when subscriber to color/image_raw/theora exists

ihadzic, 2019-11-14:
I am running the realsense ROS node and I have a subscriber to `color/image_raw/theora' topic generated by the node. If I stop the node and leave the subscriber running, I get the crash below with 100% reproduciblity. If I stop the subscriber and then start the Realsense ROS node, the crash does not occur.

``` 
[ INFO] [1573069817.666024916]: Initializing nodelet with 4 worker threads.
[INFO] [1573069817.782132]: image_snapshot: starting main loop
[ INFO] [1573069817.971005901]: RealSense ROS v2.2.6
[ INFO] [1573069817.971083050]: Running with LibRealSense v2.25.1
 06/11 14:50:18,974 WARNING [140526638655232] (types.cpp:48) hwmon command 0x4f failed. Error type: No data to return (-21).
[ INFO] [1573069819.163821614]: getParameters...
[ INFO] [1573069832.097642517]: setupDevice...
[ INFO] [1573069832.097676200]: JSON file is not provided
[ INFO] [1573069832.097737097]: ROS Node Namespace: d435
[ INFO] [1573069832.097788677]: Device Name: Intel RealSense D435I
[ INFO] [1573069832.097831581]: Device Serial No: 845112072107
[ INFO] [1573069832.097874870]: Device FW version: 05.11.11.100
5.11.11.100
[ INFO] [1573069832.097926316]: Device Product ID: 0x0B3A
[ INFO] [1573069832.097966528]: Enable PointCloud: Off
[ INFO] [1573069832.098007889]: Align Depth: Off
[ INFO] [1573069832.098047341]: Sync Mode: On
[ INFO] [1573069832.098126268]: Device Sensors:
[ INFO] [1573069832.098199575]: Stereo Module was found.
[ INFO] [1573069832.098244232]: RGB Camera was found.
[ INFO] [1573069832.098279269]: Motion Module was found.
[ INFO] [1573069832.098326009]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1573069832.098361300]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1573069832.098395512]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1573069832.112145914]: setupPublishers...
[ INFO] [1573069832.367900061]: Expected frequency for depth = 6.00000
[ INFO] [1573069836.282661578]: Expected frequency for color = 15.00000
[ INFO] [1573069840.288492998]: setupStreams...
[ INFO] [1573069840.293408716]: depth stream is enabled - width: 1280, height: 720, fps: 6
[ INFO] [1573069840.295825125]: color stream is enabled - width: 1280, height: 720, fps: 15
[ INFO] [1573069840.296194405]: insert Depth to Stereo Module
[ INFO] [1573069840.296244729]: insert Color to RGB Camera
[ INFO] [1573069840.400871752]: num_filters: 0
[ INFO] [1573069840.400916094]: SELECTED BASE:Depth, 0
[ INFO] [1573069840.401327117]: RealSense Node Is Up!
[ INFO] [1573069840.401378323]: Setting Dynamic reconfig parameters.
*** Error in `/opt/ros/kinetic/lib/nodelet/nodelet': corrupted size vs. prev_size: 0x00007fcec0258d50 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fcf104327e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x83781)[0x7fcf1043e781]
/lib/x86_64-linux-gnu/libc.so.6(realloc+0x179)[0x7fcf1043f839]
/usr/lib/x86_64-linux-gnu/libogg.so.0(oggpackB_write+0x124)[0x7fcedd99f2a4]
/usr/lib/x86_64-linux-gnu/libtheoraenc.so.1(+0x1bb1e)[0x7fcedddd8b1e]
/usr/lib/x86_64-linux-gnu/libtheoraenc.so.1(th_encode_ycbcr_in+0x3b8)[0x7fceddddad98]
/opt/ros/kinetic/lib//libtheora_image_transport.so(+0xc1de)[0x7fcede95e1de]
/opt/ros/kinetic/lib//libtheora_image_transport.so(+0xf8f2)[0x7fcede9618f2]
/opt/ros/kinetic/lib/libimage_transport.so(_ZNK15image_transport9Publisher7publishERKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEE+0x65)[0x7fcf005dc365]
/opt/ros/kinetic/lib//librealsense2_camera.so(_ZN17realsense2_camera17BaseRealSenseNode12publishFrameEN3rs25frameERKN3ros4TimeERKSt4pairI10rs2_streamiERSt3mapIS9_N2cv3MatESt4lessIS9_ESaIS7_ISA_SE_EEERKSC_IS9_NS3_9PublisherESG_SaIS7_ISA_SL_EEERKSC_IS9_S7_IN15image_transport9PublisherESt10shared_ptrINS_20FrequencyDiagnosticsEEESG_SaIS7_ISA_SW_EEERSC_IS9_iSG_SaIS7_ISA_iEEERSC_IS9_N11sensor_msgs11CameraInfo_ISaIvEEESG_SaIS7_ISA_S19_EEERKSC_IS9_NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESG_SaIS7_ISA_S1J_EEERKSC_IS8_S1J_SF_IS8_ESaIS7_IKS8_S1J_EEEb+0xc90)[0x7fcf082adbd0]
/opt/ros/kinetic/lib//librealsense2_camera.so(_ZN17realsense2_camera17BaseRealSenseNode14frame_callbackEN3rs25frameE+0x1338)[0x7fcf082b4dd8]
/opt/ros/kinetic/lib//librealsense2_camera.so(+0x6b331)[0x7fcf082b6331]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(_ZNK12librealsense12frame_source15invoke_callbackENS_12frame_holderE+0x1c4)[0x7fcef2ac1954]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(_ZN12librealsense16synthetic_source11frame_readyENS_12frame_holderE+0x33)[0x7fcef298c523]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(+0x449f9a)[0x7fcef2994f9a]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(+0x44a307)[0x7fcef2995307]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(_ZN12librealsense16processing_block6invokeENS_12frame_holderE+0x5c)[0x7fcef298cccc]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(rs2_process_frame+0x5a)[0x7fcef2a8438a]
/opt/ros/kinetic/lib//librealsense2_camera.so(_ZNSt17_Function_handlerIFvN3rs25frameEEN17realsense2_camera14PipelineSyncerEE9_M_invokeERKSt9_Any_dataOS1_+0x38)[0x7fcf082b9ad8]
/opt/ros/kinetic/lib//librealsense2_camera.so(_ZN3rs214frame_callbackISt8functionIFvNS_5frameEEEE8on_frameEP9rs2_frame+0x2d)[0x7fcf082b73ed]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(_ZNK12librealsense12frame_source15invoke_callbackENS_12frame_holderE+0x1c4)[0x7fcef2ac1954]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(+0x55df90)[0x7fcef2aa8f90]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(+0x55ec9e)[0x7fcef2aa9c9e]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(_ZN12librealsense8platform14v4l_uvc_device4pollEv+0xda9)[0x7fcef29f03b9]
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense2.so.2.25(_ZN12librealsense8platform14v4l_uvc_device12capture_loopEv+0x38)[0x7fcef29f0ca8]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xb8c80)[0x7fcf10a53c80]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7fcf0f71e6ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7fcf104c241d]
======= Memory map: ========
[... 409 lines cut ...]
7fcf11f72000-7fcf11f73000 rw-p 00004000 08:06 6422563                    /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7fcf11f73000-7fcf11fd7000 r-xp 00000000 08:06 7475530                    /opt/ros/kinetic/lib/libnodeletlib.so
7fcf11fd7000-7fcf121d6000 ---p 00064000 08:06 7475530                    /opt/ros/kinetic/lib/libnodeletlib.so
7fcf121d6000-7fcf121d9000 r--p 00063000 08:06 7475530                    /opt/ros/kinetic/lib/libnodeletlib.so
7fcf121d9000-7fcf121da000 rw-p 00066000 08:06 7475530                    /opt/ros/kinetic/lib/libnodeletlib.so
7fcf121da000-7fcf12200000 r-xp 00000000 08:06 6422703                    /lib/x86_64-linux-gnu/ld-2.23.so
7fcf12252000-7fcf123ca000 rw-p 00000000 00:00 0
7fcf123cb000-7fcf123cc000 rw-p 00000000 00:00 0
7fcf123cc000-7fcf123ea000 r-xp 00000000 08:06 6427071                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7fcf123ea000-7fcf123eb000 r--p 0001d000 08:06 6427071                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7fcf123eb000-7fcf123ec000 rw-p 0001e000 08:06 6427071                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7fcf123ec000-7fcf123ff000 rw-p 00000000 00:00 0
7fcf123ff000-7fcf12400000 r--p 00025000 08:06 6422703                    /lib/x86_64-linux-gnu/ld-2.23.so
7fcf12400000-7fcf12401000 rw-p 00026000 08:06 6422703                    /lib/x86_64-linux-gnu/ld-2.23.so
7fcf12401000-7fcf12402000 rw-p 00000000 00:00 0
7fff243ce000-7fff243ed000 rwxp 00000000 00:00 0                          [stack]
7fff243ed000-7fff243ef000 rw-p 00000000 00:00 0
7fff243f3000-7fff243f5000 r--p 00000000 00:00 0                          [vvar]
7fff243f5000-7fff243f7000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
[d435/realsense2_camera_manager-1] process has died [pid 4228, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/ros/.ros/log/bd606b2c-f370-11e9-82b3-001517c9be29/d435-realsense2_camera_manager-1.log].
log file: /home/ros/.ros/log/bd606b2c-f370-11e9-82b3-001517c9be29/d435-realsense2_camera_manager-1*.log
[d435/realsense2_camera-2] process has finished cleanly
log file: /home/ros/.ros/log/bd606b2c-f370-11e9-82b3-001517c9be29/d435-realsense2_camera-2*.log
```

doronhi, 2019-11-17:
I can't reproduce. Could you upgrade to latest realsense2_camera and latest librealsense2 and check again?
Also, what OS do you have? What launch file do you use? What computer? Any additional information that could help in reproducing this error will be appreciated.

ihadzic, 2019-11-20:
I still see the problem after latest updates. The OS is Ubuntu 16.04, 4.15.0-70. The launch file is my own derived from one of the examples that come with the realsense ROS node (relevant lines pasted below). The setup is that I have the realsense node running and I have RVIZ open. I set the fixed frame to be d435 frame (camera) and then I add `/d435/color/image_raw/theora` topic to RVIZ and the video appears. I then stop the realsense node (ctrl-c in the window where realsense node is running) and start the node again. That's the point when the node crashes. If I disable topic display in RVIZ (i.e. remove the subscriber), restart the node then, and then enable the topic in RVIZ again, everything seems fine.

```
<launch>
  ..... my other arguments not relevant for Realsense node ... 
  ....
  ...
  <arg name="HAS_DCAM" default="true" />
  <arg name="DCAM_TYPE" default="d435" />
  <group if="$(arg HAS_DCAM)">
    <group if="$(eval DCAM_TYPE.lower()=='d435')" ns="d435">
      <arg name="d435/serial_no"    default=""/>
      <arg name="d435/json_file_path" default=""/>
      <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
        <arg name="serial_no"         value="$(arg d435/serial_no)"/>
        <arg name="json_file_path"    value="$(arg d435/json_file_path)"/>
        <arg name="depth_width"       value="1280"/>
        <arg name="depth_height"      value="720"/>
        <arg name="depth_fps"         value="6"/>
        <arg name="color_width"       value="1280"/>
        <arg name="color_height"      value="720"/>
        <arg name="color_fps"         value="15"/>
        <arg name="enable_depth"      value="true"/>
        <arg name="enable_color"      value="true"/>
        <arg name="enable_infra1"     value="false"/>
        <arg name="enable_infra2"     value="false"/>
        <arg name="enable_fisheye"    value="false"/>
        <arg name="enable_gyro"       value="false"/>
        <arg name="enable_accel"      value="false"/>
        <arg name="enable_pointcloud" value="false"/>
        <arg name="enable_sync"       value="true"/>
        <arg name="tf_prefix"         value="d435"/>
      </include>
    </group>
    ....
    ..... my other nodes instantiation not relevant for Realsense
    ...
  </group>
```

doronhi, 2019-11-27:
I couldn't "add /d435/color/image_raw/theora topic to RVIZ". Instead I added "/d435/color/image_raw" as the topic of the "Image" window and set "theora" as "Transport Hint". Is that what you did?

realsense2_camera node crashed after sending a few frames. Every time. It never worked enough for me to "Ctrl-C" and restart as to reproduce the crash you describe. This did not happen when I reduced resoltion to 640x480 for both color and depth images (which is what I initially used before you added your launch file, so thanks).
It seems like a problem with the theora_image_transport plugin. I am not familiar with it.
While searching for an explanation, I found [this](https://github.com/ros-perception/image_transport_plugins/issues/4). It could explain the behavior you describe, though it offers no solution.
You could try using "compressed" instead of "theora" although it is said to be less efficient.

ihadzic, 2019-12-04:
> I couldn't "add /d435/color/image_raw/theora topic to RVIZ". Instead I added "/d435/color/image_raw" as the topic of the "Image" window and set "theora" as "Transport Hint". Is that what you did?
> 

Yes, that's essentially what makes the RVIZ a subscriber to theory subtopic under <path>/image_raw namespace

> You could try using "compressed" instead of "theora" although it is said to be less efficient.

compressed image is just a M-JPEG, which consumes about 10x more bandwidth than any motion-vector based compression (e.g. Theora or H.264) and in the network conditions that my bots run, that's prohibitively high.

So the conclusion is that this is a bug in image_transport package?

doronhi, 2019-12-05:
I am not a theora expert so but so it seems to me, that this is a theora issue.

RealSenseCustomerSupport, 2020-04-29:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-05-08 by RealSenseCustomerSupport