realsenseai/realsense-ros #992: ros can not receive messages from topic "/camera/rgb/image_rect_color"

fly-duck, 2019-11-15:
Operating System | Ubuntu
-- | --
Kernel | 4.15.0-66-generic #75~16.04.1-Ubuntu
ROS | kinetic
ROS RealSense | v2.25.0 
librealsense | 2.29.0-0~realsense0.1579 (system lib)

fly-duck, 2019-11-15:
both ros topic echo and rviz subscriber can not receive any messages

doronhi, 2019-11-17:
It is not clear to me what you mean.
First thing, what command line do you run? what kind of device?
Second, I'm not sure to what topic you try to subscribe to. Did you `rostopic list` and one of the options was "/camera/rgb/image_rect_color"? It is not a native topic. Did you remap?

fly-duck, 2019-11-18:
> It is not clear to me what you mean.
> First thing, what command line do you run? what kind of device?
> Second, I'm not sure to what topic you try to subscribe to. Did you `rostopic list` and one of the options was "/camera/rgb/image_rect_color"? It is not a native topic. Did you remap?

Thanks for your reply. The device is D435 , the command line I run is **roslaunch realsense2_camera rs_rgbd.launch** and  then **rostopic echo /camera/rgb/image_rect_color** 

Also, I have not remapped any topics and rostopic list has "/camera/rgb/image_rect_color"

fly-duck, 2019-11-18:
Here are my rostopic list 
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_infra1/camera_info
/camera/aligned_depth_to_infra1/image_raw
/camera/aligned_depth_to_infra1/image_raw/compressed
/camera/aligned_depth_to_infra1/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_infra1/image_raw/compressedDepth
/camera/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_updates
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth_registered/points
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/infra2/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressed/parameter_descriptions
/camera/infra2/image_rect_raw/compressed/parameter_updates
/camera/infra2/image_rect_raw/compressedDepth
/camera/infra2/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra2/image_rect_raw/compressedDepth/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb/image_rect_color
/camera/rgb/image_rect_color/compressed
/camera/rgb/image_rect_color/compressed/parameter_descriptions
/camera/rgb/image_rect_color/compressed/parameter_updates
/camera/rgb/image_rect_color/compressedDepth
/camera/rgb/image_rect_color/compressedDepth/parameter_descriptions
/camera/rgb/image_rect_color/compressedDepth/parameter_updates
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/rgb_rectify_color/parameter_descriptions
/camera/rgb_rectify_color/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf_static

doronhi, 2019-11-18:
The topic "/camera/rgb/image_rect_color" is not published by realsense2_camera. It is likely that you have some other plugin, such as  theora-image-transport, doing some manipulation and adding the topic.
If it is not intentional, maybe you should register to another topic? maybe `/camera/color/image_raw`? Can you describe your need from the topic?

RealSenseCustomerSupport, 2019-12-11:
----------------------------------------------

Hi @fly-duck,

Any update? Is the issue resolved?

fly-duck, 2019-12-11:
Hi @doronhi , I compiled the realsense2_camera in another environment and the topic  "/camera/rgb/image_rect_color" is not publishing. Thanks for your help.

closed 2019-12-11 by fly-duck