realsenseai/realsense-ros #994: using filters for r200

hassan-shehawy, 2019-11-15:
Hi, I had a problem with the quality of depth image I get with r200 and filtering was suggested as a solution. I'm wondering if I can tweak the filters parameters mentioned here: https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md with r200 in the launch file?
For now, I add this line to my launch file:
`<param name="filters" type="str"  value="spatial">`
However, I don't know how to set values for the filter's controls (alpha, delta, ...). When I run the reqt_reconfigure I can't see the filters 
Thanks

MartyG-RealSense, 2019-11-16:
The filters mentioned in the post-processing documentation are intended for use with cameras supported by Realsense SDK 2.0, which is not compatible with the R200.

For the R200, ROS should be launched with the R200-supporting **realsense_camera** and not realsense2_camera.

http://wiki.ros.org/realsense_camera

The R200 parameters that are dynamically reconfigurable are listed here:

http://wiki.ros.org/realsense_camera#R200_Dynamically_Reconfigurable_Parameters

hassan-shehawy, 2019-11-18:
Wow, I didn't notice there was a realsense2_camera! Sorry about that and thank you :)

closed 2019-11-18 by hassan-shehawy