realsenseai/realsense-ros #999: D435 sensor model inconsistent with datasheet

tpet, 2019-11-18:
The URDF model seems to be inconsistent with the datasheet.

In forward direction the camera_link in the URDF model is placed at the level of the bottom screw mount (relative x = 0):
https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_description/urdf/_d435.urdf.xacro#L49

The bottom screw mount position is not given in the datasheet numerically (as far as I can tell) but it seems to be in the rear half at least:
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf#page=106

According to
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf#page=64
depth origin should be only (4.2mm = 0.0042m) from the front glass, i.e., in the very front.

I'm not sure what should be the design position of the bottom screw mount to tell the correction needed.

RealSenseCustomerSupport, 2019-12-24:
----------------------------------------------

Hi @tpet

We will take a look at our datasheet to make sure we have as much data as we possibly can there.  For reference we do have an updated datasheet from the version that you reference please use the following link to the latest datasheets for all our products:  https://dev.intelrealsense.com/docs/datasheets

What I've included below is a link to the CAD models files that is available for all our camera products.  The CAD model can provide you the ability to determine distances from the bottom tripod type screw location and use that for understanding mechanical with respect to your setup and configuration.
https://dev.intelrealsense.com/docs/cad-files

Also have a look at the latest PR updates that have gone into ROS:  https://github.com/IntelRealSense/realsense-ros

RealSenseCustomerSupport, 2020-01-22:
----------------------------------------------

Hi @tpet

Do you require further assistance?  If not then the thread can be closed.

tpet, 2020-01-23:
Will you fix the URDF model? Or you don't think it's inconsistent with the datasheet?

RealSenseCustomerSupport, 2020-01-23:
----------------------------------------------

Don't think it's inconsistent.

tpet, 2020-01-23:
Let me try to explain what I think is wrong.
The issue is with the camera frames in URDF. E.g., camera_depth_optical_frame is at the same x level as the camera_link and camera_bottom_screw_frame:
https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_description/urdf/_d435.urdf.xacro#L78
which is approximately in the center (actually, more to the rear) of the sensor on x axis.
But according to the datasheet the camera frames (camera_depth_optical_frame) should be only 4.2 mm from the front side of the sensor, which is maybe 1 cm from the center on x axis.

So, if one calibrates the cameras and sets the URDF model position accordingly, the model is some 1 cm more to the front that it should be. That causes problems if one wants to e.g. filter out lidar measurements on the robot itself.

tpet, 2020-01-23:
A screenshot to illustrate:
[image]
using
```
<param name="robot_description" command="$(find xacro)/xacro --inorder '$(find realsense2_description)/urdf/test_d435_camera.urdf.xacro' use_nominal_extrinsics:=false"/>
<node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher"/>
```

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @tpet

The datasheet also outlines the X-Y information for location of depth origin in Section 4.9.1 Depth Origin X-Y Coordinates

One other question for you, are you able to get extrinsics by function call?

tpet, 2020-01-27:
Hi!

X-Y axes in the datasheet are the other two dimensions, not the one I'm concerned about. Speaking in datasheet terms: the Z coordinate is incorrect; in URDF it is actually X coordinate.

I'm using the ROS driver, I'm not using the library directly.

RealSenseCustomerSupport, 2020-02-20:
----------------------------------------------

Hi. The bottom screw mount center to front glass distance is 0.0149 as expressed in line 30 of urdf.

<xacro:property name="d435_cam_mount_from_center_offset" value="0.0149"/>

The depth origin is indeed 0.0042 from the front of glass which corresponds to infra1 frame in model.

You are right the urdf model could probably be expressed better. I'll file an issue internally for that.

Does this help answer your question?

tpet, 2020-02-21:
Well, maybe only the mesh is wrong?
If you have one more look at the picture above:
The glass is on the right.
The depth frame visualized is in the left half but it should be only 4.2 mm from the right edge, which would definitely be in the right half.
So it seems to me, that it doesn't match.

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

That is correct. The mesh is wrong. FYI...we always welcome contributions from the community as well. If you want to change the mesh we'll more than gladly look at that PR. :)

https://github.com/IntelRealSense/librealsense/blob/master/CONTRIBUTING.md

Thanks

mcamurri, 2020-03-14:
> Hi. The bottom screw mount center to front glass distance is 0.0149 as expressed in line 30 of urdf.
>
> <xacro:property name="d435_cam_mount_from_center_offset" value="0.0149"/>
>
> The depth origin is [...]

I'm sorry but I don't see the 4.2 mm offset anywhere written on the `_d435.urdf.xacro` from the latest version `2.2.13` of this package.[ I see a 0 offset instead](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_description/urdf/_d435.urdf.xacro#L47). 

In fact, when I run  `roslaunch realsense2_camera rs_d435_camera_with_model.launch` I see the `camera_link` frame is aligned with the bottom screw frame, as @tpet reported:
[image]

Now, there is no way the dynamic camera parameters can detect where is the `camera_link` from the mounting hole, so this has to be fixed in the xacro and be permanent. 

**I've made a PR:** #1124 
The 14.9 mm are used for graphical appearance only.

BTW, the front cover glass is 0.1 mm inwards from the front plate of the device. 

From CAD, the distance between the screw mount hole axis to front plate is 14.9 mm:
[image]

Therefore the correct distance from the front glass should be 14.8 mm:
[image]

Now, did you mean the front plate instead of the front glass? 

In that case, what are the 4.2 mm offset referred to? The datasheet mentions "front cover glass":
[image]
but at this stage I'm not sure.

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

Thank you for the PR  @mcamurri. Much appreciated. We'll review with engineering asap and get back to you.

Thanks

RealSenseSupport, 2020-10-07:
PR Merged.

https://github.com/IntelRealSense/realsense-ros/pull/1124

closed 2020-10-07 by RealSenseSupport