realsenseai/realsense-ros #1002: Raspberry pi 4 with two D435 not working

falkowskiUR, 2019-11-19:
----------------------------------------------------------------------------------------------------

|  Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Models                      | D435  and D435i            | 
| Firmware Versions                   | 05.10.03.00 and 05.10.13.00 | 
| Operating System & Version | Rasbian Buster  | 
| Kernel Version (Linux Only)    |  4.19.75                                        | 
| Platform                                 |Raspberry pi 4 |
| SDK Version                            |  2.30.0                        | 
|ROS verision                            |Melodic |

Hi,
After setting up all the environment on Raspberry i tested it with one D435  and it worked fine. 
I tried to connect second camera to USB 3.0 , but after launching rs_multiple_device.launch from wrapper i encounter numerous errors [logfile.txt](https://github.com/IntelRealSense/realsense-ros/files/3863157/logfile.txt). I also tested it with launching 2 separate nodes ,each for one camera (setting different Serial numbers). First camera works fine but second doesn't start at all and i get this error:

`19/11 11:01:07,552 ERROR [2711585808] (handle-libusb.h:87) failed to claim usb interface: 2, error: RS2_USB_STATUS_BUSY `
 `19/11 11:01:07,552 WARNING [2711585808] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 135 error: Device or resource busy, number: 16`

so it looks like only 1 camera can be active at a time, but I can see image from both cameras in realsense-viewer.

Is this wrapper problem, compatibility with newest SDK or maybe hardware problem? I'd appreciate any help you can provide.

MartyG-RealSense, 2019-11-19:
Intel's guide to using two cameras with ROS may provide useful insights if you have not seen it already.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

falkowskiUR, 2019-11-20:
I've seen it and used the same commands as in **Step 2: Start both cameras:** . Just like before, 1st camera works fine and 2nd doesn't start and I get error from my original comment

MartyG-RealSense, 2019-11-20:
@doronhi gave advice in the link below to another user who had a similar problem with one D435 working but not two:

https://github.com/IntelRealSense/realsense-ros/issues/432#issuecomment-414972317

falkowskiUR, 2019-11-20:
Yes i did, both cameras had different topic headers. I will try to install older version of SDK and see if it resolves the issue

samxchandlerpi, 2020-01-10:
> Yes i did, both cameras had different topic headers. I will try to install older version of SDK and see if it resolves the issue

Did you ever resolve this issue? I'm not using a Raspberry Pi, but I am facing the same while running two d435 realsense cameras. The serial numbers seem to be correct because I can run the cameras separately.

FPSychotic, 2020-01-29:
@MartyG-RealSense Did you get run RPi4 with two Realsenses on ROS? I having many of the issues you had.

MartyG-RealSense, 2020-01-29:
Hi @FPSychotic - it was @falkowskiUR who experienced the issues.  I have not used Rpi 4.

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@falkowskiUR How did you connect two cameras to Raspberry? Only one USB3 port with USB hub or two USB3 ports separately? If only one USB3 port, I would suspect if any USB bandwidth issue. Please try to turn on depth only and turn off IR1/IR2/Color which you can modify in https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/includes/nodelet.launch.xml#L24-#L29

Looking forward to your update. Thanks!

falkowskiUR, 2020-02-03:
@FPSychotic and @samxchandlerpi I'm sorry for the lack of response but i didn't resolve the issue. @RealSenseCustomerSupport I tried both approaches. First i used USB hub with external power supply, and after that I used two USB3 ports separately. 
I checked bandwidth with usbtop (with only depth frame turned on in launch file) and it was fine but it didn't fix error , so bandwidth isn't a problem.
I also tried different versions of SDK: 2.90 and 2.30 , still nothing changed.

samxchandlerpi, 2020-02-03:
> @FPSychotic and @samxchandlerpi I'm sorry for the lack of response but i didn't resolve the issue. @RealSenseCustomerSupport I tried both approaches. First i used USB hub with external power supply, [...]

Is this related to  #1051 ? As a temporary fix, I modified some of the source code and staggered the launch of the two realsense nodes. It seems like some other people had success fixing this issue by removing 'ros-kinetic-librealsense2' or downgrading realsense to 2.28.1 and realsense-ros to 2.2.8. I haven't tried these solutions yet.

falkowskiUR, 2020-02-04:
@samxchandlerpi i think that could work but i'll try it out later. Thanks for advice. For now i stopped dealing with this problem so i will close issue.

closed 2020-02-04 by falkowskiUR

JStech, 2020-02-20:
I think I'm running into this same issue, albeit not on a Raspberry Pi. I'm trying to start two D415 cameras. I've tried using `rs_multiple_devices.launch` as well as running `rs_camera.launch` twice, with different `camera` namespaces. Downgrading realsense-ros and librealsense didn't fix it for me. Here's my system info:

Info | &nbsp;
---- | -----
Camera Models | both D415
Firmware Versions | 05.12.01.00
Operating System & Version | Ubuntu 18.04
Kernel Version (Linux Only) | 5.3.0-28-generic
Platform | x86_64
SDK Version | 2.32.1
ROS verision | Melodic

When I run the cameras individually, the behavior changes depending on the order I launch the cameras. If I run them in one order, I get the same `RS2_USB_STATUS_BUSY` error, as above, but when I launch them in the other order, I get a mess of errors, including:

```
ros.realsense2_camera: RealSense Node Is Up!
 19/02 17:14:58,838 WARNING [140667653170944] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 17:14:58,898 WARNING [140667653170944] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 17:14:59,176 ERROR [140667678349056] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 19/02 17:14:59,227 WARNING [140667653170944] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 17:14:59,288 WARNING [140667653170944] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 17:14:59,374 WARNING [140667653170944] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 17:14:59,434 WARNING [140667653170944] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 17:14:59,518 WARNING [140667653170944] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/02 17:14:59,926 WARNING [140667477022464] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 23, 
ros.realsense2_camera: Hardware Notification:IR stream start failure,1.58216e+12,Error,Hardware Error
ros.realsense2_camera: Hardware Reset is needed. use option: initial_reset:=true
 19/02 17:15:01,174 ERROR [140667678349056] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 19/02 17:15:01,174 WARNING [140667678349056] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 132, error: No such device, number: 19
 19/02 17:15:01,230 WARNING [140667653170944] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 19/02 17:15:01,241 ERROR [140667485415168] (error-handling.cpp:68) Error during polling error handler: usb device disconnected
 19/02 17:15:01,302 ERROR [140666849326848] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 19/02 17:15:01,302 WARNING [140666849326848] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 130, error: No such device, number: 19
 19/02 17:15:01,334 WARNING [140667678349056] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 132 error: No such device, number: 19
 19/02 17:15:01,334 ERROR [140667678349056] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13
 19/02 17:15:01,334 WARNING [140667678349056] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 132 error: No such device, number: 19
 19/02 17:15:01,334 ERROR [140667678349056] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13
 19/02 17:15:01,335 WARNING [140667477022464] (uvc-device.cpp:644) interrupt event received: 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 [... cut ...]
 19/02 17:15:02,937 WARNING [140667678349056] (messenger-libusb.cpp:96) usb_request_cancel returned error, endpoint: 132 error: No such device, number: 19
 19/02 17:15:02,937 WARNING [140667678349056] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 132, error: No such device, number: 19
ros.realsense2_camera: Checking new devices...
 19/02 17:15:03,031 WARNING [140667460237056] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 0, 
 19/02 17:15:03,035 WARNING [140667460237056] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 0, 
ros.realsense2_camera: getParameters...
```